MKFirmware/MarlinKimbra #38: temperature hotend não muda - hotend temperature does not change

Briangsouza, 2015-04-18:
temperatura do hotend nao muda de 21 graus / hotend does not change the temperature of 21 degrees

mas o bico aquece, o mosfet é ligado / but the tip heats up, the MOSFET is connected

MagoKimbra, 2015-04-18:
please describe you configuration.
Board
and type printer

Briangsouza, 2015-04-19:
Ramps 1.4  - 33 ( fan,bed, hotness )

my printer delta

#ifndef CONFIGURATION_H
#define CONFIGURATION_H

// This configuration file contains basic settings. Select your:
//  - board type
//  - Mechanism type (cartesian-corexy-delta-scara)
//  - temperature sensor type
//
// Mechanisms-settings can be found in configuration_xxxxxx.h
// Advanced settings can be found in Configuration_adv.h

// Choose your board type.
// Either an numeric ID or name defined in boards.h is valid.
// See: https://github.com/MagoKimbra/MarlinKimbra/blob/master/Documentation/Hardware.md
#include "boards.h"
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_13_EFB
#endif

// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.
#define STRING_VERSION " 4.0.9"
#define STRING_URL "reprap.org"
#define STRING_VERSION_CONFIG_H __DATE__ "17/04/15 19:30 " __TIME__     // build date and time
#define STRING_CONFIG_H_AUTHOR "(none, default config)"   // Who made the changes.
#define STRING_SPLASH_LINE1 "v" STRING_VERSION            // will be shown during bootup in line 1
#define STRING_SPLASH_LINE2 STRING_VERSION_CONFIG_H       // will be shown during bootup in line 2

// SERIAL_PORT selects which serial port should be used for communication with the host.
// This allows the connection of wireless adapters (for instance) to non-default port pins.
// Serial port 0 is still used by the Arduino bootloader regardless of this setting.
#define SERIAL_PORT 0

// This determines the communication speed of the printer
// 115200 - 250000
#define BAUDRATE 115200

// This enables the serial port associated to the Bluetooth interface on AT90USB devices
//#define BTENABLED

// Define this to set a unique identifier for this printer, (Used by some programs to differentiate between machines)
// You can use an online service to generate a random UUID. (eg http://www.uuidgenerator.net/version4)
//#define MACHINE_UUID "00000000-0000-0000-0000-000000000000"

// If you want test the firmware uncomment below. Use Serial arduino monitor...
//#define FIRMWARE_TEST // ONLY BAUDRATE 115200

/***********************************************************************\
 **************************** Define type printer **********************
 ***********************************************************************/
//#define CARTESIAN
//#define COREXY
#define DELTA
//#define SCARA

#if defined(CARTESIAN)
  #include "Configuration_Cartesian.h"
#elif defined(COREXY)
  #include "Configuration_Corexy.h"
#elif defined(DELTA)
  #include "Configuration_Delta.h"
#elif defined(SCARA)
  #include "Configuration_Scara.h"
#endif
/***********************************************************************/
[... 661 lines cut ...]

// MOVEMENT SETTINGS
#define HOMING_FEEDRATE {100*60, 100*60, 100*60, 0}      // set the homing speeds (mm/min)

// default settings
// delta speeds must be the same on xyz
#define DEFAULT_AXIS_STEPS_PER_UNIT {80,80,80,582.05,625,625,625}        // X, Y, Z, E0, E1, E2, E3
#define DEFAULT_MAX_FEEDRATE {300,300,300,45,100,100,100}             // X, Y, Z, E0, E1, E2, E3 (mm/sec)
#define DEFAULT_RETRACTION_MAX_FEEDRATE {150,150,150,150}             // E0, E1, E2, E3 (mm/sec)
#define DEFAULT_MAX_ACCELERATION {3000,3000,3000,10000,1000,1000,1000} // X, Y, Z, E0, E1, E2, E3 maximum start speed for accelerated moves.

#define DEFAULT_ACCELERATION          600      // X, Y, Z and E max acceleration in mm/s^2 for printing moves
#define DEFAULT_RETRACT_ACCELERATION  9000      // X, Y, Z and E max acceleration in mm/s^2 for retracts
#define DEFAULT_TRAVEL_ACCELERATION   3000      // X, Y, Z acceleration in mm/s^2 for travel (non printing) moves

// Offset of the extruders (uncomment if using more than one and relying on firmware to position when changing).
// The offset has to be X=0, Y=0 for the extruder 0 hotend (default extruder).
// For the other hotends it is their distance from the extruder 0 hotend.
//#define HOTEND_OFFSET_X {0.0, 5.00, 0.0, 0.0} // (in mm) for each extruder, offset of the hotend on the X axis
//#define HOTEND_OFFSET_Y {0.0, 5.00, 0.0, 0.0} // (in mm) for each extruder, offset of the hotend on the Y axis

// The speed change that does not require acceleration (i.e. the software might assume it can be done instantaneously)
#define DEFAULT_XYJERK 45  // (mm/sec)
#define DEFAULT_ZJERK  45  // (mm/sec)
#define DEFAULT_EJERK   45  // (mm/sec)

Briangsouza, 2015-04-19:
ok, i fix!! ;) my pins.h. I had changed

closed 2015-04-19 by Briangsouza