MKFirmware/MarlinKimbra #44: Configurazioni sulla SD

simone97, 2015-05-07:
Giusto per curiosità come mai è stato tolto il sistema per le configurazioni sulla SD?
Saluti,
Simone

MagoKimbra, 2015-05-07:
L'ho fatto sulla eeprom, avevo problemi con Arduino due. Poi in questo modo possono averlo tutti anche chi non ha la sd...

--- Messaggio originale ---

Inviata: 07 Maggio 2015 15:05
A: "MagoKimbra/MarlinKimbra" <MarlinKimbra@noreply.github.com>
Oggetto: [MarlinKimbra] Configurazioni sulla SD (#44)

https://github.com/MagoKimbra/MarlinKimbra/issues/44

simone97, 2015-05-07:
Che genere di problemi??
Comuque se lo fai sulla eeprom la uccidi.
Puoi leggere/scrivere la eeprom un massimo di 100 000 volte
se tu calcoli che una persona che usa abbastanza frequentemente la stampante la tiene accesa anche 5 ore al giorno, e che l'aggiormanento della eeprom avverrebe ogni minuto per un totale di 300 volte al giorno
Ottieni che dopo un anno hai già bello che superato quel numero di cicli e ciao eeprom :-1:

MagoKimbra, 2015-05-07:
A questo non ci avevo pensato... Ho visto che la makerbot ha un sistema simile...  Forse perché loro scrivono solo a fine stampa....Ok allora aggiusto e rimetto la scrittura su SD....

Date: Thu, 7 May 2015 06:54:42 -0700

simone97, 2015-05-07:
Bé già a fine stampa é accettabile, tu calcola che con 3 volte al giorno due una decina d'anni.
Io l'ho fatto su SD proprio per non distruggere la eeprom.
Io avevo pensato poi di dividere i due sistemi, cioé chi vuole usa la eeprom a suo rischio e pericolo, chi ha la SD invece usa quella.
Chi abilita la eeprom ha le configurazioni sulla eeprm come ora, e a fine stampa vengono aggiornate le eventuali statistiche.
Mentre chi abilita la SD ha tutte le configurazioni sulla SD, con la possibilità di essere aggiornate in live senza Dover andare nel menu per cliccare Salva Configurazioni e in più le statistiche aggiornate ogni minuto.

PS: Se riaggiungi il fatto dell'sd mettilo nel branch di sviluppo che mi é venuto in mente ora che le configurazioni vengono salvate anche durante la stampa in corso in caso che la stampa non sia partita dalla SD stessa e questo potrebbe causare blocchi durante la stampa tramite host.
Basta aggiungere un piccolo controllo comunque.
Saluti,
Simone

https://github.com/MagoKimbra/MarlinKimbra/issues/44#issuecomment-100011228

simone97, 2015-05-09:
Fammi sapere quando lo hai ri-aggiunto che ci lavoro su

MagoKimbra, 2015-05-25:
Sul branch development ho rimesso tutto com'era...

Date: Sat, 9 May 2015 10:51:34 -0700

simone97, 2015-05-27:
perfetto questo fine settimana trovo il tempo di sistemarlo

closed 2015-05-28 by MagoKimbra