MKFirmware/MarlinKimbra #46: Extruder sometimes spins backwards on first turn

phizz166, 2015-06-21:
I recently got MarlinKimbra running on my MendelMax 3 (RAMBO) with a custom dual extruder setup. I have one 3mm geared extruder and one 1.75mm direct drive, with two different types of motor -- impossible to handle with standard Marlin. So far, with MarlinKimbra, it works very well most of the time. (add this to your list of supported hardware!). 

There is one occasional bug I can't figure out. This usually happens after switching a tool with T1/T0 commands. Sometimes after the switch, when I try to manually extrude filament (I use Repetier-Host), the extruder spins backwards and retracts instead. I'll click "extrude 10mm" and instead it retracts 10mm. I can do this a number of times and the extruder will retract every time. Then, if I click the "retract" button, the extruder *does* retract whatever distance it's supposed to...and after that, the directions are back to normal. Both extrusion and retraction behave as intended.

So basically, after switching a tool the extrusion direction is reversed (but not the retraction), until a retraction command is sent, at which time it solves itself.

This hasn't caused major problems with prints so far because every tool change includes a retraction command, so it may be covering up the problem. I haven't gotten really deep into the calibration of toolchange behavior so I don't know for sure.

Any ideas on what might be going on? Are there logs I could pull to help solve the problem?

phizz166, 2015-06-21:
This looks like it may be related to issue #5 as well, though I haven't experienced that specific problem yet.

MagoKimbra, 2015-06-22:
I apologize for my bad English ...
In the dev version, I have already patched this ...
Today I put it in the version 4.1.2 you can try.
Thanks ..

closed 2015-08-19 by MagoKimbra

Astroman, 2017-02-06:
I'm having this exact problem on another Marlin branch. Would you please elaborate on the fix so I can report it elsewhere? Thanks.