MKFirmware/MarlinKimbra #47: Full Graphics LCD Panels LCD Contrast error

percarsten, 2015-06-25:
When selecting any full Graphics LCD 128 x 64 with Azteeg X3 Pro it errors during compilation
Errors are
Error at line 517 in file configuration_store.cpp

same error doesnt matter what controller board is selected....basically anything where contrast can be adjusted

\Kimbra Master\MarlinKimbra-master\MarlinKimbra\configuration_store.cpp: In function 'void Config_ResetDefault()':
Error at line 517 in file configuration_store.cpp
\Kimbra Master\MarlinKimbra-master\MarlinKimbra\configuration_store.cpp:517: error: 'lcd_contrast' was not declared in this scope
Failed compiling sketch

MagoKimbra, 2015-06-25:
Sorry.. I fix... Try please..

percarsten, 2015-06-25:
ok partially fixed
Error at line 23 in file configuration_store.cpp:516
Kimbra Master\MarlinKimbra-master\MarlinKimbra\configuration_store.cpp:516:23: error: #if with no expression
Failed compiling sketch

it looks like this is in cine 516 configuration store cpp

MagoKimbra, 2015-06-25:
Sorry... One moment..

percarsten, 2015-06-26:
Ok I tried it again
I am still having a few issues
I am running this on an AzteegX3 Pro and Viki2 LCD Controller
The encoder button does not select any menu
I had to add definitions for DOGLCD_A0 and CS or it would not compile

closed 2015-06-26 by percarsten