MKFirmware/MarlinKimbra #52: Delta dual extruder Bug

Dougwar, 2015-08-05:
I noticed that Marlin has one bug for dual extruder in deltas

please check these links:

http://forums.reprap.org/read.php?178,300903

https://groups.google.com/forum/#!topic/deltabot/foGyIqCf2bs

Thanks

Dougwar, 2015-08-13:
did you read this?

MagoKimbra, 2015-08-14:
Yes i read this... But i do not understand the problem..
I have tested on my MarlinKimbra and not see problem..

Dougwar, 2015-08-14:
Thanks for the answer.
I'm trying to print with dual extruders in my delta, when the tool change T0 to T1 or any change, the hotend go up and lost reference, it go up for every tool change. If you still don't understand I can make a video with the problem, just let me know.
Thanks

MagoKimbra, 2015-08-14:
Ok i retest this..

MagoKimbra, 2015-08-16:
I retest on my delta and haven't problem...
Please send me your configuration.h and configuration_delta.h

closed 2016-04-25 by Dougwar