MKFirmware/MarlinKimbra #55: Errore Grammaticale

simonepri, 2015-08-10:
Nel sito web:
http://marlinkimbra.it/
nella sezione Funzioni Addizionali
Power Consumation Sensor
va corretto con
Power Consumption Sensor

MagoKimbra, 2015-08-10:
Ok aggiustato prova a verificare che è tutto ok tu che ce l'hai!!

Date: Mon, 10 Aug 2015 13:23:46 -0700

http://marlinkimbra.it/

simonepri, 2015-08-11:
Appena torno a Roma ti faccio sapere ma dovrebbe andare.
Comunque avevo intenzione di implementare un semplice comando M?? che permetta di ripristinare gli assi nel caso di perdita di step.
Ipotizziamo che durante la stampa ci si accorga in tempo che la stampante abbia perso degli step il comando in questione effettuerebbe l'homing degli assi e poi continuerebbe la stampa nel modo corretto.

Per le cartesiane gli assi che possono perdere step sono solo X e Y quindi basterebbe fare l'homing dei due assi e poi riprendere la stampa senza toccare l'asse Z, quindi piuttosto semplice.
L'unico problema é che non conosco bene la meccanica delle altri stampanti e purtroppo dove sono la connessione internet non mi permette di navigare in modo decente quindi non posso documentarmi.
Sapresti dirmi come sarebbe possibile farlo per i vari tipi di meccanica?

Dato che ho qualche giorno libero se hai altre cose in mente che possano essere utili dimmelo che provo a lavorarci.
Saluti,
Simone

MagoKimbra, 2015-08-11:
Non ho mica capito perché ti serva un nuovo comando...Se ti accorgi che ha perso i passi fai pausa su repetier fai il g28 x0 y0 e riparti...

Date: Tue, 11 Aug 2015 07:26:03 -0700

simonepri, 2015-08-11:
Per rapidità e poi ad esempio l'homing dei singoli assi non é presente nei menú dell'lcd.
Poi scusa ad esempio sulla delta non puoi fare semplicemente l'homing.
Se non ricordo male l'homing delle delta é al min, pertanto se fai l'homing distruggi la stampa (o mi sbaglio?).

https://github.com/MagoKimbra/MarlinKimbra/issues/55#issuecomment-129954366

MagoKimbra, 2015-08-11:
No, la Delta ha l'homing sui max, infatti è l'unica che può fare homing di tutto gli assi senza sbattere su alcun pezzo...

Date: Tue, 11 Aug 2015 09:36:59 -0700

https://github.com/MagoKimbra/MarlinKimbra/issues/55#issuecomment-129954366

MagoKimbra, 2015-08-11:
Adesso che ci penso c'è una cosa che volevo fare da tempo, ma non ci ho mai messo su la testa, visto che con la sd ci sai fare vediamo se ti stuzzica sta cosa...Salvare tutti i parametri del fw su SD in un file che si può richiamare per riconfigurare la scheda, oppure per poter dare il fw precompilato con il file di configurazione che si richiama quando si accende la scheda...

From: magokimbra@hotmail.com
To: reply@reply.github.com
Subject: RE: [MarlinKimbra] Errore Grammaticale (#55)
Date: Tue, 11 Aug 2015 19:40:59 +0200

No, la Delta ha l'homing sui max, infatti è l'unica che può fare homing di tutto gli assi senza sbattere su alcun pezzo...

Date: Tue, 11 Aug 2015 09:36:59 -0700

https://github.com/MagoKimbra/MarlinKimbra/issues/55#issuecomment-129954366

simonepri, 2015-08-11:
Ci avevo già pensato di farlo però essenzialmente non ha molto senso.
Innanzitutto non tutti hanno l'SD, in secondo luogo spesso si accende la stampante anche senza SD quindi sarebbe un po una rottura avere le configurazioni sull'SD e dipendere da essa.
Io trovo che le configurazioni nel FW siano l'idea migliore.
Al massimo la cosa comoda sarebbe poter fare un file che in fase di compilazione viene letto e vengono abilitate/cambiate le varie funzioni/impostazioni, così ogni volta basta solo buttare lo stesso file nella cartella e compilare.

Comunque io invece stavo pensando anche di aggiungere la possibilità di interrompere la stampa e riprenderla in un secondo momento (quindi poter, ad esempio, spegnere la stampante e continuare il giorno dopo). Non so magari a chi fa stampe grosse può tornare utile.

Ovviamente solo per le stampe da SD.
Il parser per i file di configurazione tanto l'ho giá fatto quindi semplicemente aggiungo una voce al menu che permette di interrompere la stampa e che salva su un file: il nome del file stampato e la riga del ultimo gcode eseguito su un file.
Nel menu comparirà quindi una nuova voce "riprendi ultima stampa".
La voce scompare (e quindi il record viene cancellato dal file) o se viene ripresa la stampa o se l'utente avvia una nuova stampa ex novo.
Mi sembra semplice e funzionale anche se non so quanto effettivamente possa incidere sulla qualità di stampa il fatto di interrompere la stampa.

https://github.com/MagoKimbra/MarlinKimbra/issues/55#issuecomment-129954366

---
https://github.com/MagoKimbra/MarlinKimbra/issues/55#issuecomment-130049489

closed 2015-08-12 by MagoKimbra