MKFirmware/MarlinKimbra #56: REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER Wrong beeper pin

KriLL3, 2015-08-12:
It's set to 64, in stock marlin it's set to 37, 37 works, 64 doesn't, took me seconds to fix it, but might as well be correct by default.

MagoKimbra, 2015-08-12:
Which baord?

KriLL3, 2015-08-12:
Specifically it's  this one: http://eud.dx.com/product/robotale-3d-ramps-1-4-3d-control-board-set-lcd12864-2560-r3-3d-1-4-4988-844315282#.VcuPJ7Ukodw but it's a clone of this one: http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller in stock marlin you uncomment the 560 line in configuration.h "#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER" I looked through pins.h and found mention of "#define BEEPER 37" under "REPRAP_DISCOUNT_SMART_CONTROLLER" which seems to apply due to

"#if defined (REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER)
 #define DOGLCD
 #define U8GLIB_ST7920
 #define REPRAP_DISCOUNT_SMART_CONTROLLER
#endif"

But changing "#define BEEPER_PIN" to 37 doesn't work for me, I assumed it would but no it doesn't, I can't send M300 beeps and the LCD doesn't make any sounds when buttons are pushed, not a big deal but I used beeps to tell me the state of the machine and when it's done.

Is it because Marlin uses "BEEPER" and MarlinKimbra uses "BEEPER_PIN" perhaps? Then again for some reason a lot of pins don't seem to line up for some reason but still works 
Marlin:
"        #define BTN_EN1 31
        #define BTN_EN2 33
        #define BTN_ENC 35"

MarlinKimbra:
"  #define BTN_EN1             14
  #define BTN_EN2             16
  #define BTN_ENC             17"

I would have thought arduino pins are arduino pins or is there some swapping going on? It's a RAMPs1.4 on an arduino mega 2560.

MagoKimbra, 2015-08-14:
I test Ramps 1.4 width REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER and function all perfectly...
Do you test width 4.1.2 or 4.1.4 dev?

simonepri, 2015-12-27:
The pins are right.
If you need more support open a new issue.

closed 2015-12-27 by simonepri