MKFirmware/MarlinKimbra #60: Preblemi cinetica cartesiana

simonepri, 2015-08-19:
Dalla versione 4.1.3 in su ho problemi con i movimenti della stampante, credo sia dovuto all'implementazione dell' EJerk sull estrusore oppure all'aggiunta dell'accellerazione sulla ritrazione.
Spiego il problema.
Avvio il gcode dalla SD, la stampante si riscalda, fa homing e tutto, si muove verso il punto per iniziare la stampa, fa un rumore duro (tipo il tonfo quando si spegne uno stepper) e parte la stampa.
Nel gcode io ho le accellerazioni per i movimenti settate a 1250
Se le alzo a 3000 la stampante non fa proprio uscire il filamento durante la stampa (tipo dryrun).

Tutto ciò non succede con la versione 4.1.2.
Credo sia un bug significativamente grave, ora ispezioni i cambiamenti nel codice tra le versioni e vedo di fixare.
Perchè anche la qualità di stampa ne risente molto.
Lo stesso gcode a causa di questo problema con la 4.1.2 viene fuori molto meglio di quello fatto con la 4.1.5

simonepri, 2015-08-19:
Sto isolando il problema
Fino ai Commit del
Jun 15, 2015
Il bug non cè quindi il problema non è ne l'accellerazione della E ne il EJerk

Nei Commit del
Aug 09, 2015
C'è il bug

Ora trovo il commit incriminato e risolvo :+1:

simonepri, 2015-08-19:
Dopo aver fleshato 10 FW ho trovato da dove parte il BUG:
Jun 25, 2015 (Return to Delay for High Low Step)
https://github.com/MagoKimbra/MarlinKimbra/commit/8f9f6e0d042bc36edc2b13eea91956bd6c6e77c0
Qual'è l'intento delle modifiche che hai fatto??

Commits on Jun 21, 2015 (FIx step_loops) NO BUG
Commits on Jun 22, 2015 (Same fix miss step!) NO BUG
Commits on Jun 25, 2015 (Fix LCD contrast) NO BUG
Commits on Jun 25, 2015 (Return to Delay for High Low Step) BUG

MagoKimbra, 2015-08-19:
Tu sei un genio... Grazie infinite.....
il problema è qui in conditional.h#if defined(CONFIG_STEPPERS_TOSHIBA) || !defined(ENABLE_HIGH_SPEED_STEPPING)      #define MAX_STEP_FREQUENCY 10000 // Max step frequency for Toshiba Stepper Controllers      #define DOUBLE_STEP_FREQUENCY MAX_STEP_FREQUENCY    #else      #define MAX_STEP_FREQUENCY 40000 // Max step frequency for Arduino mega      #define DOUBLE_STEP_FREQUENCY 10000    #endif
Quando ho ripristinato mi sono dimenticato di toglier il controllo || !defined(ENABLE_HIGH_SPEED_STEPPING)Quindi è vera la prima parte e non setta bene le frequenze degli stepper...

Date: Wed, 19 Aug 2015 09:19:49 -0700

8f9f6e0

simonepri, 2015-08-19:
Ok risolto, ora pubblico una pull per il fix
Il BUG affliggeva solo le stampanti con l'arduino Mega

simonepri, 2015-08-19:
Si ci ero arrivato anche io è la prima cosa che ho controllato stavo solo testando per scaramanzia nel caso ci fosse altro ahah sembra funzionare

closed 2015-08-19 by simonepri