MKFirmware/MarlinKimbra #70: Steps per unit

trublu289, 2015-09-13:
I am using two extruders and the steps per unit set both extruders with the value for second extruder and I have to manually set the steps for the first extruder on each restart. Is there a option to display this in eeprom settings repetier host as well?

MagoKimbra, 2015-09-14:
M503 read value in EEPROM
M502 set value from default
M501 set value from EEPROM
M500 Store value in EEPROM

trublu289, 2015-09-14:
I changed the first value then the second value changed to the first 1. And
no matter how many times I flash it doesn't change. Some how they are
different even though the first and second are the same. Not sure now.

MagoKimbra, 2015-09-14:
Please explain to me the commands to change the steps. And which version of firmware you are using.

trublu289, 2015-09-14:
I'm using the latest version of kimbra and I have settings pre set in
firmware but I can only change the first extruder steps by going into
eeprom settings in repetier host.

MagoKimbra, 2015-09-14:
A ok.. Do not use repetier host window for change value in EEPROM, because it use only one valor for all extruder.. If you write M503 on command text you read the command gcode for change any parameter. Example for step per unit the commandi is M92 Xxx Yyy Zzz Eee for another extruder use command T, example M92 T1 Eee for change step extruder 1, M92 T2 Eee for change extruder 2 etc etc...

trublu289, 2015-09-14:
Makes sense why the weird effects now.  Is there a way to have independent
fan control for each extruder?.

MagoKimbra, 2015-09-14:
You mean the fans to cool the hotend or pla?

trublu289, 2015-09-14:
Pla

simonepri, 2015-12-27:
@trublu289 At the moment only one fan is supported.
But you can use EXTRUDER_AUTO_FAN feature in order to archive almost the same result.
If you need more support open a new issue.

closed 2015-12-27 by simonepri