MKFirmware/MarlinKimbra #77: não compila #define Z_PROBE_ENDSTOP

Briangsouza, 2015-10-01:
Arduino: 1.6.4 (Mac OS X), Placa:"Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from base.h:52:0,
                 from blinkm.cpp:6:
sanitycheck.h:1519: error: #error DEPENDENCY ERROR: You must set Z_PROBE_PIN to a valid pin if you enable Z_PROBE_ENDSTOP
       #error DEPENDENCY ERROR: You must set Z_PROBE_PIN to a valid pin if you enable Z_PROBE_ENDSTOP
        ^
#error DEPENDENCY ERROR: You must set Z_PROBE_PIN to a valid pin if you enable Z_PROBE_ENDSTOP

  Este relatório deveria ter mais informações
  "Mostrar saída verbosa durante a compilação"
  habilitado em Arquivo > Preferências.

pins.h:

/****************************************************************************************
* 33
* RAMPS 1.3 / 1.4
* RAMPS_13_HFB (Hotend0, Fan, Bed)
****************************************************************************************/

#if MB(RAMPS_13_HFB)
  #define KNOWN_BOARD 1
  
  #if !defined(__AVR_ATmega1280__) && !defined(__AVR_ATmega2560__)
    #error Oops!  Make sure you have 'Arduino Mega' selected from the 'Tools -> Boards' menu.
  #endif
  
  #define LARGE_FLASH true
  
  //X axis pins
  #define ORIG_X_STEP_PIN         54
  #define ORIG_X_DIR_PIN          55
  #define ORIG_X_ENABLE_PIN       38
  #define X_MIN_PIN                3
  #define X_MAX_PIN                2
  
  //Y axis pins
  #define ORIG_Y_STEP_PIN         60
  #define ORIG_Y_DIR_PIN          61
  #define ORIG_Y_ENABLE_PIN       56
  #define Y_MIN_PIN               14
  #define Y_MAX_PIN               15
  
  #define Y2_STEP_PIN             36
  #define Y2_DIR_PIN              34
  #define Y2_ENABLE_PIN           30

  //Z axis pins
  #define ORIG_Z_STEP_PIN         46
  #define ORIG_Z_DIR_PIN          48
  #define ORIG_Z_ENABLE_PIN       62
  #define Z_MIN_PIN               18
  #define Z_MAX_PIN               19
  #define Z_PROBE_PIN             18

  #define Z2_STEP_PIN             36
  #define Z2_DIR_PIN              34
  #define Z2_ENABLE_PIN           30

  //E axis pins
  #define ORIG_E0_STEP_PIN        26
  #define ORIG_E0_DIR_PIN         28
  #define ORIG_E0_ENABLE_PIN      24

  #define ORIG_E1_STEP_PIN        36
  #define ORIG_E1_DIR_PIN         34
  #define ORIG_E1_ENABLE_PIN      30
  
  #define SDPOWER                 -1
  #define SDSS                    53
  #define LED_PIN                 13
  
  #define ORIG_FAN_PIN             9
  #define PS_ON_PIN               12

MagoKimbra, 2015-10-02:
You must define Z_PROBE_PIN in configuration_pins.h

#if ENABLED(Z_PROBE_ENDSTOP)
  #define Z_PROBE_PIN -1
#endif

Briangsouza, 2015-10-05:
ok!

closed 2015-12-27 by simonepri