MKFirmware/MarlinKimbra #79: Now to connect Nextion HMI LCD to RAMPS from Kossel Delta 3D Printer ?

fred2088, 2015-10-26:
Dear Sir,

How to connect Nextion HMI LCD to RAMPS from Kossel Delta 3D Printer ?
it's seem connected to TX1 and RX2 which is same as Z MAX and Zmin ?

Regards

MagoKimbra, 2015-10-26:
You can easily move the pin with the online configurator.

Date: Mon, 26 Oct 2015 09:08:27 -0700

fred2088, 2015-10-27:
Dear Sir,

Thanks for the info,

How to update the the nextion hmi lcd in order to work with the marlin ?

I have change the marlin to use nextion hmi lcd and Serial2 and uploaded to MEGA2560
tx connect to rx2
rx connect to tx2
the hmi lcd shows the demo screen

Regards

From: MagoKimbra 
Sent: Tuesday, October 27, 2015 3:37 AM
To: MagoKimbra/MarlinKimbra 
Cc: fred 
Subject: Re: [MarlinKimbra] Now to connect Nextion HMI LCD to RAMPS from Kossel Delta 3D Printer ? (#79)

You can easily move the pin with the online configurator.

Date: Mon, 26 Oct 2015 09:08:27 -0700

MagoKimbra, 2015-10-27:
You must download the firmware into the display ... You find it in the folder nextion is the file .tft or files .hmi if you use the editor nextion ....
For download read Nextion itead documentation...

fred2088, 2015-10-27:
Thanks for the info,

But can’t open the HMI file using lastest version 0.3 editor
and also got restore file error after copy the tft file to sd card

Regards

From: MagoKimbra 
Sent: Tuesday, October 27, 2015 10:28 PM
To: MagoKimbra/MarlinKimbra 
Cc: fred 
Subject: Re: [MarlinKimbra] Now to connect Nextion HMI LCD to RAMPS from Kossel Delta 3D Printer ? (#79)

You must download the firmware into the display ... You find it in the folder nextion is the file .tft or files .hmi if you use the editor nextion ....
For download read Nextion itead documentation...

fred2088, 2015-11-04:
Dear Sir, 
I changed nexconfig.h to Serial2
and connect D16 to tx and D17 to RX
I can upload the firmware to the HMI LCD and it can display  the five pluzze logo.

the serial debug shows the following :
------------------------------------------------
echo: echo: MarlinKimbra V4; MarlinKimbra for 4 extruder
echo: Last Updated: Nov  4 2015 21:46:05 | Author: (none, default config)
echo: Compiled: Nov  4 2015
echo: Free Memory: 3588 PlannerBufferBytes: 1232
ok Hardcoded SD Default Settings Loaded
echo: Hardcoded Default Settings Loaded
echo: setup_for_endstop_move > enable_endstops(true)
echo: SD init fail
--------------------
is the hmi lcd show info screen if successful connected ?

Regards

From: MagoKimbra 
Sent: Tuesday, October 27, 2015 3:37 AM
To: MagoKimbra/MarlinKimbra 
Cc: fred 
Subject: Re: [MarlinKimbra] Now to connect Nextion HMI LCD to RAMPS from Kossel Delta 3D Printer ? (#79)

You can easily move the pin with the online configurator.

Date: Mon, 26 Oct 2015 09:08:27 -0700

MagoKimbra, 2015-11-04:
Do you set Nextion in configuration_feature.h?

fred2088, 2015-11-05:
Yes, enabled Nextion from header file

Regards

From: MagoKimbra 
Sent: Thursday, November 05, 2015 5:43 AM
To: MagoKimbra/MarlinKimbra 
Cc: fred 
Subject: Re: [MarlinKimbra] Now to connect Nextion HMI LCD to RAMPS from Kossel Delta 3D Printer ? (#79)

Do you set Nextion in configuration_feature.h?

MagoKimbra, 2015-11-05:
Have you in log "Nextion LCD not connected!" or "Nextion LCD connected!"?

If you haven't this, the nextion system aren't active...

fred2088, 2015-11-05:
where to find this log message ? from serial log ?

I have just un-comment the Nextion line

From: MagoKimbra 
Sent: Thursday, November 05, 2015 3:04 PM
To: MagoKimbra/MarlinKimbra 
Cc: fred 
Subject: Re: [MarlinKimbra] Now to connect Nextion HMI LCD to RAMPS from Kossel Delta 3D Printer ? (#79)

Have you in log "Nextion LCD not connected!" or "Nextion LCD connected!"?

If you haven't this, the nextion system aren't active...

fred2088, 2015-12-08:
dear sir,

is there any debug message after connected to Nextion HMI LCD ?
I can’t found out it’s connected or not.

TIA
Regards
Fred

From: MagoKimbra 
Sent: Thursday, November 05, 2015 3:04 PM
To: MagoKimbra/MarlinKimbra 
Cc: fred 
Subject: Re: [MarlinKimbra] Now to connect Nextion HMI LCD to RAMPS from Kossel Delta 3D Printer ? (#79)

Have you in log "Nextion LCD not connected!" or "Nextion LCD connected!"?

If you haven't this, the nextion system aren't active...

fred2088, 2015-12-08:
enabled Nextion from configuration_feature.h already

// Nextion HMI panel
// REMEMBER TO INSTALL Nextion library in your ARDUINO library folder. You can find it in Arduino\libraries\
#define NEXTION

start
echo: echo: MarlinKimbra V4; MarlinKimbra for 4 extruder
echo: Last Updated: Dec  8 2015 12:59:41 | Author: (Tinyboy, default config)
echo: Compiled: Dec  8 2015
echo: Free Memory: 3588 PlannerBufferBytes: 1232
ok Hardcoded SD Default Settings Loaded
echo: Hardcoded Default Settings Loaded
echo: setup_for_endstop_move > enable_endstops(true)
echo: SD card ok

From: MagoKimbra 
Sent: Thursday, November 05, 2015 3:04 PM
To: MagoKimbra/MarlinKimbra 
Cc: fred 
Subject: Re: [MarlinKimbra] Now to connect Nextion HMI LCD to RAMPS from Kossel Delta 3D Printer ? (#79)

Have you in log "Nextion LCD not connected!" or "Nextion LCD connected!"?

If you haven't this, the nextion system aren't active...

fred2088, 2015-12-10:
Try to download the firmware again, got the below log, and already #define  NEXTION from configuration_feature
but no log message about HMI LCD ?

Regards
-----------------------------------------------------------------
start
echo: echo: MarlinKimbra V4; MarlinKimbra for 4 extruder
echo: Last Updated: Dec 10 2015 11:13:54 | Author: (none, default config)
echo: Compiled: Dec 10 2015
echo: Free Memory: 5442 PlannerBufferBytes: 1232
ok Hardcoded SD Default Settings Loaded
echo: Hardcoded Default Settings Loaded
echo: Steps per unit:
echo:   M92 X80.00 Y80.00 Z3200.00 E625.00
echo: Maximum feedrates (mm/s):
echo:   M203 X300.00 Y300.00 Z2.00 E100.00
echo: Maximum Acceleration (mm/s2):
echo:   M201 X3000 Y3000 Z50 E1000

echo: Accelerations: P=printing, V=travel and T* R=retract
echo:   M204 P3000.00 V3000.00
echo:   M204 T0 R10000.00
echo: Advanced variables: S=Min feedrate (mm/s), V=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:   M205 S0.00 V0.00 B20000 X10.00 Z0.40 E5.00
echo: Home offset (mm):
echo:   M206 X0.00 Y0.00 Z0.00
echo: PID settings:
echo:   M301 H0 P40.00 I7.00 D60.00
echo: Filament settings: Disabled
echo: Power on time:
echo:   0 days 0 hours 0 minutes
echo: setup_for_endstop_move > enable_endstops(true)
wait
wait

MagoKimbra, 2015-12-10:
Please attach your file configuration_overall and configuration_feature....

fred2088, 2015-12-10:
Dear Sir,

File is attached...
Thanks

[Configuration_Feature.h.txt](https://github.com/MagoKimbra/MarlinKimbra/files/58104/Configuration_Feature.h.txt)
[Configuration_Overall.h.txt](https://github.com/MagoKimbra/MarlinKimbra/files/58105/Configuration_Overall.h.txt)

fred2088, 2015-12-10:
1. #if ENABLED(NEXTION)
2.  #include <Nextion.h>
3. #endif
it's seems #include <Nextion.h> is not compiled even I uncomment #define NEXTION, the sketches compiles without any error
-------------------------------------------------
when i try to comment  line 1 and 3 to force compilation of line 2, I got compile error 
C:\Users\User\AppData\Local\Temp\buildc8537909c2afc015b00072723ace1c92.tmp/core\core.a(HardwareSerial0.cpp.o): In function `__vector_25':

C:\Program Files\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial0.cpp:48: multiple definition of `__vector_25'

sketch\MarlinSerial.cpp.o:sketch/MarlinSerial.cpp:54: first defined here

c:/program files/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

collect2.exe: error: ld returned 1 exit status

fred2088, 2016-01-01:
Sorry to have the same error again after downloading the marlinkimbra.zip again on 01/01/2016
with enabled #define NEXTION

Regards
---------------------------------------------------------------------------------------------------------------------------------
C:\Users\User\AppData\Local\Temp\buildb61f730cdf244d32938116ddca3401ae.tmp/core\core.a(HardwareSerial0.cpp.o): In function `__vector_25':

C:\Program Files\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial0.cpp:48: multiple definition of `__vector_25'

sketch\MarlinSerial.cpp.o:sketch/MarlinSerial.cpp:54: first defined here

c:/program files/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

collect2.exe: error: ld returned 1 exit status

exit status 1

fred2088, 2016-01-01:
I can get connected to Nextion HMI LCD and have some command work by showing from Log file
echo: Power on time:
echo:   0 days 0 hours 0 minutes
echo: Nextion LCD connected!
echo: enqueueing "G91"
echo: enqueueing ""
echo: enqueueing "G90"
ok 

change MarlinSerial.cpp as follow , allow me to compile the marlin and upload to mega 2560

#if defined(M_USARTx_RX_vect)
  // fixed by Mark Sproul this is on the 644/644p
//  SIGNAL(SIG_USART_RECV)
//  SIGNAL(M_USARTx_RX_vect) {
//    unsigned char c  =  M_UDRx;
//    store_char(c);
//  }
#endif

I can now go futher to test the nextion hmi lcd for debuging over 2 months
Thanks all

closed 2016-01-02 by fred2088

roscher123, 2017-02-27:
Hi All im new at all this I have a nextion4.3 that I want to connect to my ramps 1.4 I have moved the pins already. im how ever not sure where to connect the ground and the volt power on the ramps1.4 im using pin 16 and 17 TX1 and RX2 for coms.

MagoKimbra, 2017-02-27:
Use V+ e gnd on endstop pin...

roscher123, 2017-02-27:
hi not working the screen just flashes

roscher123, 2017-02-27:
Thx Mang maybe you can give me a step by step from the start pls