MKFirmware/MarlinKimbra #84: NEXTION HMI - NexVar issues

jgauchia, 2015-11-28:
Hi, 
First of all, good job with your firmware compatibility with NEXTION HMI! (and sorry for my English).

I'm testing this feature but I've some issues reading variable values like vacmd , etc...

In setmovePopCallback function of nextion_lcd.cpp gets the vacmd value and put it into buffer for a enqueued command when pushed move buttons.
This value doesn't match with value returned by NEXTION in the IDE program, it's something like this:

echo: enqueueing "G91"
echo: enqueueing "n feedrate (mm/s), V=Min trav"   <------------ seems a incorrect reading¿? old buffer data¿?
echo: enqueueing "G90"

I've tried to clean the buffer, create a new buffer, etc... but only gets the same result or garbage...

It's normal? How can I test it (enable debugging maybe)?

Thank you!!!

MagoKimbra, 2015-11-29:
uhm this is very strange ...
I check this tomorrow on my nextion lcd...

jgauchia, 2015-11-29:
Hi MagoKimbra
Thank you!

Also, I'll make a little program to test connection and reading variables..

jgauchia, 2015-12-04:
I've tried some code changes but still doesn't work.
The communication between Nextion and Ramps is correct.

I've seen at reprap forum that a user (who write a lot of posts) has the same problem , I think.

I'll continue testing...

jgauchia, 2015-12-22:
Hi,

I've made some tests with no good results.
Last test, without EPROM in firmware returns this:

echo: enqueueing "G91"
echo: enqueueing "Maximum Acceleration (mm/s2):"
echo: enqueueing "G90"
echo: Unknown command: "Maximum Acceleration (mm/s2):"

It's seems a buffer reading error.
However fan control and SD reading was correct... 

I'll continue testing.

jgauchia, 2016-02-01:
Any news about it? :(

MagoKimbra, 2016-02-01:
No sorry.. I haven't time for chek this... I hope this week...

jgauchia, 2016-02-01:
Ah! No problem MagoKimbra. ! I'm still upgrading my printer

jgauchia, 2016-03-28:
Updated.
With new version of MarlinKimbra & Pronterface

Display something like this.
**
Echo: enqueueing "G91"
Attempted to write invalid text to console, which could be due to an invalid baudrate
Echo: enqueueing "G90"
Attempted to write invalid text to console, which could be due to an invalid baudrate
[ERROR] Traceback (most recent call last):
File "/home/jordi/Printrun/printrun/printcore.py", line 118, in logError
try: self.errorcb(error)
File "/home/jordi/Printrun/printrun/pronsole.py", line 257, in logError
msg = u"".join(unicode(i) for i in msg)
File "/home/jordi/Printrun/printrun/pronsole.py", line 257, in
msg = u"".join(unicode(i) for i in msg)
File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0x8f in position 25: invalid start byte**

jgauchia, 2016-03-28:
The same command with Repetier

13:44:36.887 : Echo: enqueueing "G91"
13:44:36.891 : Echo: enqueueing "?%d+$K?"4?!?
13:44:36.893 : ??"
13:44:36.893 : Echo: enqueueing "G90"
13:44:36.899 : Error: Unknown command: "?%d+$K?"4?!?

jgauchia, 2016-05-29:
Still continue with this version

When touch something in nextion screen:

Echo:Nextion LCD connected!
Error:SD init fail
Echo:enqueueing "G91"
Echo:enqueueing "o high: "
Echo:enqueueing "G90"
Error:Unknown command: "o high: 
Echo:enqueueing "G91"
Echo:enqueueing "o high: "
Echo:enqueueing "G90"

Seems the string:

_#define SERIAL_STEPPER_TOO_HIGH                 "Steprate too high: "_

carmelf, 2016-08-09:
hello jgauchia 
any luck with this problem ? i have the same thing

jgauchia, 2016-08-10:
Hi carmelf
No, I'm using Marlin now. I don't know if MarlinKimbra works . I've finished my printer this week and started to use Marlin. Maybe , when my printer works at 100% , I'll try MarlinKimbra again.

closed 2026-10-03 by jgauchia