MKFirmware/MarlinKimbra #85: Reprapdiscount LCD panel: SD card not functional + Bed temperature not displaying

Nutshell1, 2015-12-11:
I recently upgraded my Mendel(i2) to a dual extrusion setup and marlin desperately lacks functionality like same steps/mm and pid settings for both extruders. 

The latest Marlin release has seamless LCD panel setup which shows extruder and heated bed temps and has full SD card functionality. But the dual extrusion issues remain.

After messing around with MarlinX2 with highly disappointing results, I decided to give this a shot and it is a beautiful piece of code. However, I have some issues:
-I just cannot get the SDcard to work or heat bed temp to display. 
-Also the preheat menu lacks the preheat bed only option. 
-The X and Y positions are kind of overkill since during print all that I kind of care about is %completion or to a minor degree the Z height

My current setup is a Mendel(i2), bowden fed dual extruder(greg-wade herringbone gears), all metal hotends (http://www.ebay.com/itm/181910932202?_trksid=p2060353.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT), RAMBO board, Marlin firmware(https://github.com/MarlinFirmware/Marlin)- using the september release.

While I agree I might be asking for too much, wirefree SD card printing has kind of spoilt me. I would love it, if someone could bridge these 2 firmwares.

MagoKimbra, 2015-12-12:
I'm sorry but i do not understand... Do you use Marlin or MarlinKimbra??

explaining your problem well...

Nutshell1, 2015-12-12:
Currently Im still on Marlin but I really want to switch to MarlinKimbra

MagoKimbra, 2015-12-13:
Ok.. But i do not understand your problem! Please explain me...

Nutshell1, 2015-12-14:
All my problems are about the LCD panel as follows:
-I cannot use the SD card with the LCD panel (it does not read the card)
-LCD panel does not display bed temperature
-Preheat on LCD does not have bed preheat

I am using Reprapdiscount LCD panel

MagoKimbra, 2015-12-15:
Do you have set SD CARD?
Do you have version 4.1.2 STABLE or 4.2.2 dev?

simonepri, 2015-12-21:
@Nutshell1 
1) You are right i've already fixed it, this afternoon i'll realease a fix.
2) I'll check for you
3) This is a consequence of problem 2, bed preaheat is supported.

MartinsZB, 2015-12-23:
I am using similar config Reprapdiscount LCD with my Prusa I3. Just switched from Marlin to MarlinKimbra (dev. 4.2) with two extruders
1. SD card is working
2. LCD show both hotend temperature and bod temperature OK

By the way, thanks for great firmware!

simonepri, 2015-12-27:
Let me know if you still have problems with:
https://github.com/MagoKimbra/MarlinKimbra/tree/dev

closed 2016-03-03 by MagoKimbra