MKFirmware/MarlinKimbra #86: Configurator tool & Power off

MartinsZB, 2015-12-23:
Hi! I just switched from Marlin, as I am preparing for my new Cyclope hotend. Now I am using Prusa I3 with two Bowden extruders, two E3D hotends, RAMPS 1.4 board and RepRapDiscount Smart controller LCD. MarlinKimbra 4.2. There is two things I think would be nice to see:
1. In Configurator tool - I miss the distance settings for two hotend setup. (X&Y from hotend0 to hotend1)
2. In LCD menu I missed "Power Off/On" option, what just sends M81/M80. As I am writing, I think, that is probably missing, because I choose PowerSupply0? I am using this function to shutdown my OctoPi. It is watching for M81 and if it is find - Shutting down the Raspberry.

simonepri, 2015-12-27:
Power off/on is already supported and available.
You should set
POWER_SUPPLY
to 1 or 2.
See Configuration_Basic.h

MartinsZB, 2015-12-30:
Yes, thanks - POWER_SUPPLY 1 is working.

MagoKimbra, 2015-12-30:
Ok... Thank's

closed 2015-12-30 by MagoKimbra