MKFirmware/MarlinKimbra #88: M105 answer

MartinsZB, 2015-12-30:
Hi! I am running dual head printer with MarlinKimbra and Octoprint. Octoprint is confused, when receiving response on M105 code:
Recv: T:77.1 /205.0 B:59.5 /60.0 T0:206.5 /205.0 T1:77.1 /205.0 B@:0 @:127 @0:40 @1:127 W:?

This is the answer - can you please explain, what the "B@:0 @:127 @0:40 @1:127 W:?" part is about?
Thanks,
Martins

MagoKimbra, 2015-12-30:
is the output value on the fet.
B@ it is to the bed, while @ the dell'hotend active.
@0 is hotend 0 e @1 hotend 1

Octoprint do not accept this information?

MartinsZB, 2015-12-30:
Wait. Here is the response:
Recv: T:77.1 /205.0 B:59.5 /60.0 T0:206.5 /205.0 T1:77.1 /205.0 B@:0 @:127 @0:40 @1:127 W:?

T:77.1/205.0 - it is actual t and set t for active extruder
B59.5/60.0 - it is actual t and set t for heated bed
T0:206.5 /205.0 - it is actual t and set t for extruder0
T1:77.1 /205.0 - it is actual t and set t for extruder1
So
B@:0 -  means bed - what is @:127?
@0 means hotend 0 - what the :40 mean and also :127 after @1?
W:? - in my opinion must be stabilization time

Thanks!

MagoKimbra, 2015-12-30:
B@:0 is output value on fet for BED
@:127 is output value on fet active extruder
@0:40 is output value on fet for Extruder 0
@1:127 is output value on fet for Extruder 1

For W:? i don't understand why you have this... 
Which version do you have by MK??

MartinsZB, 2015-12-31:
I am using 4.2.2 dev version (downloaded from online config tool)

MagoKimbra, 2015-12-31:
Ok i check this...

closed 2016-01-13 by MagoKimbra