MKFirmware/MarlinKimbra #91: New Nextion Display Interface

Deepoff, 2016-01-14:
Good day! I want to give You to use Nextion the screen interface I designed for my printer. If You are interested in such cooperation, please contact me on deepoff@ya.ru. Besides, I have a few code modifications for marlin to work the best initialization Nextion display.

link to china Nextion clone project (and program) in archive:
https://yadi.sk/d/22hPbiSpmuyqn

MagoKimbra, 2016-01-14:
Good day, we are happy to help, and you help me ...I'm not very good with the graphics ...I have seen your changes and are fine ...Only I have difficulty using the editor in your language .. 😉What changes did you marlin?
Date: Wed, 13 Jan 2016 17:23:31 -0800

https://yadisk/d/22hPbiSpmuyqn

Deepoff, 2016-01-14:
Отлично! Я могу, к примеру, нарисовать Вам такой же проект, под оригинальный Nextion дисплей, но иконки будут такого же размера (заново их нарисовать могу, но лень). С Вами хочу адаптировать код Marlin под иерархию этого проекта дисплея. Также могу помочь с переводом на русский язык, если это нужно.

Deepoff, 2016-01-14:
void lcd_init() {
//    delay(1000);
//    NextionON = nexInit();
    while ((NextionON = nexInit()) != true){
      delay(100);
    }

This code works better than the old one when initializing the display. With legacy code, my display was not able to boot, and did not respond to clicking (not working with the controller)

MagoKimbra, 2016-01-21:
good. Only a problem if the nextion came off the card locks them.
I would do a series of five tests and if not found out saying it is disconnected ..

Deepoff, 2016-01-22:
indeed, it would be better! Write your email so I can write you suggestions, or contact me at my email

closed 2016-03-03 by MagoKimbra