MKFirmware/MarlinKimbra #93: Arduino Compiler problem

giacomobonatti, 2016-01-29:
All the time that try to compile MarlinKimbra.ino (downloaded the end of configuration) I receive the following compiler error:

In file included from /Configuration.h:613,
                 from /Marlin.h:25,
                 from blinkm.cpp:5:
Configuration_adv.h:335: error: expected ',' or ';' before 'const'

how can i fix it? i have also an older (i think) version of MarlinKimbra.ino and it works

MagoKimbra, 2016-01-29:
Che if the line const unsigned int dropsegments=5; have the ; to end line!!!

giacomobonatti, 2016-01-29:
Yes, it has

Inviato da iPhone

MagoKimbra, 2016-01-29:
and here const bool FILRUNOUT_PIN_INVERTING = true;?

giacomobonatti, 2016-01-30:
The code is right.. I have also tryed to download the code from another PC with another Arduino versione and the error was the same 

Inviato da iPhone

MagoKimbra, 2016-01-30:
Attach your configuration.h

giacomobonatti, 2016-01-30:
i have attached 2 files:
Configuration.h
Configuration_adv.h

Because the error for my arduino compiler is in configuration_adv.h

---
Questa e-mail è stata controllata per individuare virus con Avast antivirus.
https://www.avast.com/antivirus

#ifndef CONFIGURATION_H
#define CONFIGURATION_H

// This configuration file contains basic settings. Select your:
//  - board type
//  - Mechanism type (cartesian-corexy-delta-scara)
//  - temperature sensor type
//
// Mechanisms-settings can be found in configuration_xxxxxx.h
// Advanced settings can be found in Configuration_adv.h

#include "boards.h"
// Choose your board type.
// Either an numeric ID or name defined in boards.h is valid.
// See: https://github.com/MagoKimbra/MarlinKimbra/blob/master/Documentation/Hardware.md

#define MOTHERBOARD BOARD_RAMPS_13_HFF

// User-specified version info of this build to display in [Pronterface, etc] terminal window during
// startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
// build by the user have been successfully uploaded into firmware.
#define STRING_VERSION "4.1.2"
#define STRING_URL "reprap.org"
#define STRING_VERSION_CONFIG_H __DATE__ " " __TIME__     // build date and time
#define STRING_CONFIG_H_AUTHOR "(none, Prusa R2 EVO)"      // Who made the changes.
#define STRING_SPLASH_LINE1 "v" STRING_VERSION            // will be shown during bootup in line 1
#define STRING_SPLASH_LINE2 STRING_VERSION_CONFIG_H       // will be shown during bootup in line 2

// SERIAL PORT selects which serial port should be used for communication with the host.
// This allows the connection of wireless adapters (for instance) to non-default port pins.
// Serial port 0 is still used by the Arduino bootloader regardless of this setting.
#define SERIAL_PORT 0

// This determines the communication speed of the printer
// 115200 - 250000
#define BAUDRATE 250000

// This enables the serial port associated to the Bluetooth interface on AT90USB devices
//#define BTENABLED

// Define this to set a unique identifier for this printer, (Used by some programs to differentiate between machines)
// You can use an online service to generate a random UUID. (eg http://www.uuidgenerator.net/version4)
//#define MACHINE_UUID "00000000-0000-0000-0000-000000000000"

// If you want test the firmware uncomment below. Use Serial arduino monitor...
//#define FIRMWARE_TEST // ONLY BAUDRATE 115200

/***********************************************************************\
 **************************** Define type printer **********************
 ***********************************************************************/
#define CARTESIAN
//#define COREXY
//#define DELTA
//#define SCARA
/***********************************************************************\

/***********************************************************************\
 ********************** Do not touch this section **********************
 ***********************************************************************/
#if defined(CARTESIAN)
[... 1454 lines cut ...]
  #define E0_MAX_CURRENT 1000  //in mA
  #define E0_SENSE_RESISTOR 91 //in mOhms
  #define E0_MICROSTEPS 16     //number of microsteps
  
//  #define E1_IS_TMC
  #define E1_MAX_CURRENT 1000  //in mA
  #define E1_SENSE_RESISTOR 91 //in mOhms
  #define E1_MICROSTEPS 16     //number of microsteps 
  
//  #define E2_IS_TMC
  #define E2_MAX_CURRENT 1000  //in mA
  #define E2_SENSE_RESISTOR 91 //in mOhms
  #define E2_MICROSTEPS 16     //number of microsteps 
  
//  #define E3_IS_TMC
  #define E3_MAX_CURRENT 1000  //in mA
  #define E3_SENSE_RESISTOR 91 //in mOhms
  #define E3_MICROSTEPS 16     //number of microsteps   

#endif

/******************************************************************************\
 * enable this section if you have L6470  motor drivers. 
 * you need to import the L6470 library into the arduino IDE for this
 ***********************************

MagoKimbra, 2016-01-30:
In this part...
const bool FILRUNOUT_PIN_INVERTING = false haven't ; to end line... Please write ; and compile..

I change the configurator online for this problem...

//===========================================================================
//============================= Filament Runout Sensor ======================
//===========================================================================
#define FILAMENT_RUNOUT_SENSOR // Uncomment for defining a filament runout sensor such as a mechanical or opto endstop to check the existence of filament
                                 // It is assumed that when logic high = filament available
                                 //                    when logic  low = filament run out
#ifdef FILAMENT_RUNOUT_SENSOR
const bool FILRUNOUT_PIN_INVERTING = false      // Should be uncommented and true or false should assigned
  #define ENDSTOPPULLUP_FIL_RUNOUT            // Uncomment to use internal pullup for filament runout pins if the sensor is defined.
#define FILAMENT_RUNOUT_SCRIPT "M600"      // Script execute when filament run out
#endif

closed 2016-03-03 by MagoKimbra