MKFirmware/MarlinKimbra #98: error compiling project

prophetmaster, 2016-02-15:
"C:\Program Files (x86)\Arduino/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10708 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" -I"C:\Program Files (x86)\Arduino\libraries\LiquidCrystal\src" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire" -I"C:\Program Files (x86)\Arduino\libraries" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries" -I"D:\visual studio\Common7\IDE\Extensions\opqmsfqs.42m\Micro Platforms\default\debuggers" -I"C:\Users\Sullivan\Documents\Arduino\libraries" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\utility" "C:\Users\Sullivan\AppData\Local\V.Micro\Arduino\Builds\MK\mega_atmega2560\MK.cpp" -o "C:\Users\Sullivan\AppData\Local\V.Micro\Arduino\Builds\MK\mega_atmega2560\MK.cpp.o"
MK.ino:In file included from
base.h:14:27: fatal error: module\macros.h: No such file or directory
:#include "module\macros.h"
:compilation terminated
"C:\Program Files (x86)\Arduino/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10708 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" -I"C:\Program Files (x86)\Arduino\libraries\LiquidCrystal\src" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire" -I"C:\Program Files (x86)\Arduino\libraries" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries" -I"D:\visual studio\Common7\IDE\Extensions\opqmsfqs.42m\Micro Platforms\default\debuggers" -I"C:\Users\Sullivan\Documents\Arduino\libraries" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\utility" "C:\Users\Sullivan\AppData\Local\V.Micro\Arduino\Builds\MK\mega_atmega2560\Configuration_Store.cpp" -o "C:\Users\Sullivan\AppData\Local\V.Micro\Arduino\Builds\MK\mega_atmega2560\Configuration_Store.cpp.o"
Configuration_Store.cpp:In file included from
base.h:14:27: fatal error: module\macros.h: No such file or directory
:#include "module\macros.h"
:compilation terminated
Error compiling project sources

MagoKimbra, 2016-02-15:
https://github.com/MagoKimbra/MarlinKimbra/blob/master/Documentation/Compilation.md

You must have 1.6.7 Arduino IDE.....

prophetmaster, 2016-02-17:
yes work with arduino software but not with visual studio... bug in visual
studio...

-- 

[image: Agence-Akira] <http://www.agence-akira.ch>

[image: Facebook] <https://www.facebook.com/buchs.sullivan> [image: Google
Plus] <https://plus.google.com/u/0/+SullivanBuchs> [image: Linkedin]
<http://%20%20%20%20https://ch.linkedin.com/pub/sullivan-buchs/30/2a6/87b>

Sullivan Buchs / Directeur
+41792551903

Agence-Akira
Route de Diesse 8 2518 Nods
http://www.agence-akira.ch

This e-mail message may contain confidential or legally privileged
information and is intended only for the use of the intended recipient(s).
Any unauthorized disclosure, dissemination, distribution, copying or the
taking of any action in reliance on the information herein is prohibited.
E-mails are not secure and cannot be guaranteed to be error free as they
can be intercepted, amended, or contain viruses. Anyone who communicates
with us by e-mail is deemed to have accepted these risks. Company Name is
not responsible for errors or omissions in this message and denies any
responsibility for any damage arising from the use of e-mail. Any opinion
and other statement contained in this message and any attachment are solely
those of the author and do not necessarily represent those of the company.

MagoKimbra, 2016-03-03:
MK is compatibile width Arduino IDE, sorry!!!

closed 2016-03-03 by MagoKimbra