MKFirmware/MarlinKimbra #99: Error compiling if extruder count == 0

pixel3design-hub, 2016-02-17:
As I want to use MK for Laser CNC Engraver/Cutter, I don't need the extruder stepper driver or extruder heater.
I need to configure like that:
X & E0 => dual X drivers
Y & E1 => Dual Y drivers
Z => Single z Driver

I'll use pins 42 & 44 for TTL PWM and Laser ON/OFF.

But when I turn to 0 the extruder count, Arduino throw me an error.

> // This defines the number of extruder real or virtual
>#define EXTRUDERS 0
>// This defines the number of Driver extruder you have and use
>#define DRIVER_EXTRUDERS 0

In file included from sketch\module\../base.h:37:0,
                 from sketch\module\MK_Main.cpp:31:
sketch\module\../module/conditionals.h:574:39: error: too many initializers for 'int [0]'
   #define ARRAY_BY_EXTRUDERS(v1) { v1 }
                                       ^
sketch\module\MK_Main.cpp:71:38: note: in expansion of macro 'ARRAY_BY_EXTRUDERS'
 int extruder_multiplier[EXTRUDERS] = ARRAY_BY_EXTRUDERS(100);
                                      ^
sketch\module\../module/conditionals.h:574:39: error: too many initializers for 'int [0]'
   #define ARRAY_BY_EXTRUDERS(v1) { v1 }
                                       ^
sketch\module\MK_Main.cpp:72:37: note: in expansion of macro 'ARRAY_BY_EXTRUDERS'
 int density_multiplier[EXTRUDERS] = ARRAY_BY_EXTRUDERS(100);
                                     ^
sketch\module\../module/conditionals.h:574:39: error: too many initializers for 'float [0]'
   #define ARRAY_BY_EXTRUDERS(v1) { v1 }
                                       ^
sketch\module\MK_Main.cpp:74:34: note: in expansion of macro 'ARRAY_BY_EXTRUDERS'
 float filament_size[EXTRUDERS] = ARRAY_BY_EXTRUDERS(DEFAULT_NOMINAL_FILAMENT_DIA);
                                  ^
sketch\module\../module/conditionals.h:574:39: error: too many initializers for 'float [0]'
   #define ARRAY_BY_EXTRUDERS(v1) { v1 }
                                       ^
sketch\module\MK_Main.cpp:75:42: note: in expansion of macro 'ARRAY_BY_EXTRUDERS'
 float volumetric_multiplier[EXTRUDERS] = ARRAY_BY_EXTRUDERS(1.0);
                                          ^
In file included from sketch\module\../module/conditionals.h:230:0,
                 from sketch\module\../base.h:37,
                 from sketch\module\MK_Main.cpp:31:
Pins.h:5581: error: 'E0_STEP_PIN' was not declared in this scope
 #define _E0_PINS E0_STEP_PIN, E0_DIR_PIN, E0_ENABLE_PIN, HEATER_0_PIN, analogInputToDigitalPin(TEMP_0_PIN),
                  ^
sketch\module\../module/../Pins.h:5604:25: note: in expansion of macro '_E0_PINS'
                         _E0_PINS _E1_PINS _E2_PINS _E3_PINS \
                         ^
sketch\module\MK_Main.cpp:95:30: note: in expansion of macro 'SENSITIVE_PINS'
 const int sensitive_pins[] = SENSITIVE_PINS; ///< Sensitive pin list for M42
                              ^
Pins.h:5581: error: 'E0_DIR_PIN' was not declared in this scope
 #define _E0_PINS E0_STEP_PIN, E0_DIR_PIN, E0_ENABLE_PIN, HEATER_0_PIN, analogInputToDigitalPin(TEMP_0_PIN),
                               ^
sketch\module\../module/../Pins.h:5604:25: note: in expansion of macro '_E0_PINS'
                         _E0_PINS _E1_PINS _E2_PINS _E3_PINS \
                         ^
sketch\module\MK_Main.cpp:95:30: note: in expansion of macro 'SENSITIVE_PINS'
 const int sensitive_pins[] = SENSITIVE_PINS; ///< Sensitive pin list for M42
                              ^
Pins.h:5581: error: 'E0_ENABLE_PIN' was not declared in this scope
 #define _E0_PINS E0_STEP_PIN, E0_DIR_PIN, E0_ENABLE_PIN, HEATER_0_PIN, analogInputToDigitalPin(TEMP_0_PIN),
                                           ^
sketch\module\../module/../Pins.h:5604:25: note: in expansion of macro '_E0_PINS'
                         _E0_PINS _E1_PINS _E2_PINS _E3_PINS \
                         ^
sketch\module\MK_Main.cpp:95:30: note: in expansion of macro 'SENSITIVE_PINS'
 const int sensitive_pins[] = SENSITIVE_PINS; ///< Sensitive pin list for M42
                              ^
exit status 1
'E0_STEP_PIN' was not declared in this scope

Can You provide a configuration for CNC works?
Thanks and keep up the good mood.

MagoKimbra, 2016-02-17:
Put still extruders and extruder drivers to 1 then do not use it. Nothing happens..

pixel3design-hub, 2016-02-17:
OK.
And about duplicate stepper signals?

closed 2016-02-24 by pixel3design-hub