MKFirmware/MarlinKimbra #103: Nextion serial garbage

vandykl, 2016-03-08:
If I touch the splash screen after startup I get the following on the serial communication of Simplify3D and similarly Pronterface:

RECEIVED: External Reset
RECEIVED: MK_4.2.7_dev
RECEIVED: Last Updated: Mar  8 2016 11:57:21 | Author: (none, default config)
RECEIVED: Compiled: Mar  8 2016
RECEIVED: Free Memory: 3895 PlannerBufferBytes: 1232
RECEIVED: ok Hardcoded SD Default Settings Loaded
  Connected to machine!
RECEIVED: Echo: Hardcoded Default Settings Loaded
Hardcoded Default Settings Loaded
RECEIVED: Config: Steps per unit:
RECEIVED: Config:   M92 X80.00 Y80.00 Z3200.00 E625.00
RECEIVED: Config: Maximum feedrates (mm/s):
SENT: T0
RECEIVED: Config:   M203 X300.00 Y300.00 Z2.00 E100.00
RECEIVED: Config: Maximum Acceleration (mm/s2):
RECEIVED: Config:   M201 X3000 Y3000 Z50 E1000
RECEIVED: 
RECEIVED: Config: Accelerations: P=printing, V=travel and T* R=retract
RECEIVED: Config:   M204 P3000.00 V3000.00
RECEIVED: Config:   M204 T0 R10000.00
RECEIVED: Config: Advanced variables: S=Min feedrate (mm/s), V=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
RECEIVED: Config:   M205 S0.00 V0.00 B20000 X10.00 Z0.40 E5.00
RECEIVED: Config: Home offset (mm):
RECEIVED: Config:   M206 X0.00 Y0.00 Z0.00
RECEIVED: Config: PID settings:
RECEIVED: Config:   M301 H
RECEIVED: Config: Filament settings: Disabled
RECEIVED: Info: Power on time:
RECEIVED: Info:   0 days 0 hours 0 minutes
RECEIVED: Info: Filament printed:
RECEIVED: Info:   0 Km 0 m 0 cm 0 mm
RECEIVED: Echo: Nextion LCD connected!
Nextion LCD connected!
RECEIVED: ok 
SENT: M105
RECEIVED: ok T:189.3 /0.0 @:0
SENT: M105
RECEIVED: ok T:189.0 /0.0 @:0
SENT: M105
RECEIVED: ok T:188.6 /0.0 @:0
RECEIVED: Error: '<ÿÚøÅWÞ9Ü;üíWï¾=ïé¯º9g¿]ºvÃ7üÿ¬w5?^ïíù_n{_ä[áîõ`Y¹wÆúïöüÚ¥ðýïgÖÃó·>óÊ×3:ãØÎ~ü¿
WARNING: CRC mismatch detected
RECEIVED: lÛOWø¼Üìê/9sClÊ_ýö³ü þNï~øÏ{¿?}ØÌ}Îó¥ùj|Ö>ÏÇ1m×5\ç=ýæö[z<~÷õñágãÑ=vzUümùÝ|òs½Uüaîé\?{ÖïkÊoøÙæ]º·Í¯ó¯÷ìví#¥îæÍRbæÙ}Ï¨a9õ---, system stopped! Heater_ID: 0
RECEIVED: Error: '<ÿÚøÅWÞ9Ü;üíWï¾=ïé¯º9g¿]ºvÃ7üÿ¬w5?^ïíù_n{_ä[áîõ`Y¹wÆúïöüÚ¥ðýïgÖÃó·>óÊ×3:ãØÎ~ü¿
WARNING: CRC mismatch detected
RECEIVED: lÛOWø¼Üìê/9sClÊ_ýö³ü þNï~øÏ{¿?}ØÌ}Îó¥ùj|Ö>ÏÇ1m×5\ç=ýæö[z<~÷õñágãÑ=vzUümùÝ|òs½Uüaîé\?{ÖïkÊoøÙæ]º·Í¯ó¯÷ìví#¥îæÍRbæÙ}Ï¨a9õ---, system stopped! Heater_ID: 0
SENT: M105
SENT: M105
SENT: M105

jgauchia, 2016-03-15:
I've a same issue , when touch the screen

start
External Reset
MK_4.2.8_dev
Last Updated: Mar 15 2016 17:50:36 | Author: (ICEMAN, default config)
Compiled: Mar 15 2016
Free Memory: 1792 PlannerBufferBytes: 1232
ok Hardcoded SD Default Settings Loaded
Echo: Hardcoded Default Settings Loaded
Config: Steps per unit:
Config:   M92 X158.40 Y200.05 Z5165.80 E199.42
Config:   M92 T1 E199.42
Config: Maximum feedrates (mm/s):
Config:   M203 X300.00 Y250.00 Z3.00 E300.00
Config:   M203 T1 E300.00
Config: Maximum Acceleration (mm/s2):
Config:   M201 X200 Y200 Z50 E3000
Config:   M201 T1 E3000

Config: Accelerations: P=printing, V=travel and T* R=retract
Config:   M204 P2500.00 V3000.00
Config:   M204 T0 R10000.00
Config:   M204 T1 R10000.00
Config: Advanced variables: S=Min feedrate (mm/s), V=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Config:   M205 S0.00 V0.00 B20000 X10.00 Z0.40 E5.00
Config:   M205 T1 E5.00
Config: Home offset (mm):
Config:   M206 X0.00 Y0.00 Z0.00
Config: Hotend offset (mm):
Config:   M218 T0 X0.00 Y0.00 Z0.00
Config:   M218 T1 X0.00 Y0.00 Z0.00
Config: Z Probe offset (mm)
Config:   M666 P-26.00
Config: PID settings:
Config:   M301 HConfig:   M301 H P13.14 I0.85 D50.47
Config: Filament settings: Disabled
Info: Power on time:
Info:   0 days 0 hours 0 minutes
Info: Filament printed:
Info:   0 Km 0 m 0 cm 0 mm
Echo: Nextion LCD connected!
Error: SD init fail
[ERROR] Error: SD init fail

Echo: Home X/Y abans Z
SENDING:G0 X120.0 Y82.5 F5000
SENDING:G0 X120.0 Y82.5 F5000
Error: N, system stopped! Heater_ID: 0
[ERROR] Error: N, system stopped! Heater_ID: 0

Error: N, system stopped! Heater_ID: 0
[ERROR] Error: N, system stopped! Heater_ID: 0

Error: N, system stopped! Heater_ID: 0
[ERROR] Error: N, system stopped! Heater_ID: 0

Error: N, system stopped! Heater_ID: 0
[ERROR] Error: N, system stopped! Heater_ID: 0

Error: N, system stopped! Heater_ID: 0
[ERROR] Error: N, system stopped! Heater_ID: 0

Error: N, system stopped! Heater_ID: 0
[ERROR] Error: N, system stopped! Heater_ID: 0

jgauchia, 2016-03-15:
And when I try to move axis with touch:

Echo: enqueueing "G90"
[ERROR] Traceback (most recent call last):
  File "/home/jordi/Printrun/printrun/printcore.py", line 118, in logError
    try: self.errorcb(error)
  File "/home/jordi/Printrun/printrun/pronsole.py", line 257, in logError
    msg = u"".join(unicode(i) for i in msg)
  File "/home/jordi/Printrun/printrun/pronsole.py", line 257, in <genexpr>
    msg = u"".join(unicode(i) for i in msg)
  File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xa8 in position 25: invalid start byte

Attempted to write invalid text to console, which could be due to an invalid baudrate

jgauchia, 2016-03-16:
More tests...
When firmware starts and touch nextion screen the temperature of extruder/s increase quicky and the printer doesn't works (hanged).
If the screen was in splash screen the printer works without problems.

MagoKimbra, 2016-03-17:
I tried, but in no case I get problems like yours.
I just can not understand why you have those problems ..

jgauchia, 2016-03-17:
Hi.

The configuration overall?

MagoKimbra, 2016-03-17:
Yes

jgauchia, 2016-03-17:
/**
 * Configuration_Overall.h
 * Here you can define all your custom settings and they will overwrite configurations in the main configuration files.
 */

/***********************************************************************
 ********************** Serial comunication type ***********************
 ***********************************************************************/
#define SERIAL_PORT 0
#define BAUDRATE 115200
#define STRING_CONFIG_H_AUTHOR "(ICEMAN, default config)"
#define MACHINE_UUID "00000000-0000-0000-0000-000000000000"
/***********************************************************************/

/***********************************************************************
 *************************** Board type ********************************
 ***********************************************************************/
#define MOTHERBOARD BOARD_RAMPS_13_HHF
/***********************************************************************/

/***********************************************************************
 *************************** Mechanism type ****************************
 ***********************************************************************/
#define MECHANISM MECH_CARTESIAN
/***********************************************************************/

/***********************************************************************
 ************************** Extruders number ***************************
 ***********************************************************************/
#define EXTRUDERS 2
#define DRIVER_EXTRUDERS 2
/***********************************************************************/

/***********************************************************************
 **************************** Power supply *****************************
 ***********************************************************************/
#define POWER_SUPPLY 0
//#define PS_DEFAULT_OFF
/***********************************************************************/

/***********************************************************************
 **************************** Thermistor type **************************
 ***********************************************************************/
#define TEMP_SENSOR_0 11
#define TEMP_SENSOR_1 11
/***********************************************************************/

/***********************************************************************
 ************************* Temperature limits **************************
 ***********************************************************************/
#define HEATER_0_MINTEMP 5
#define HEATER_0_MAXTEMP 275
#define HEATER_1_MINTEMP 5
#define HEATER_1_MAXTEMP 275
#define PLA_PREHEAT_HOTEND_TEMP 210
#define PLA_PREHEAT_HPB_TEMP 0
#define PLA_PREHEAT_FAN_SPEED 255
#define ABS_PREHEAT_HOTEND_TEMP 240
#define ABS_PREHEAT_HPB_TEMP 100
#define ABS_PREHEAT_FAN_SPEED 255
#define GUM_PREHEAT_HOTEND_TEMP 230
#define GUM_PREHEAT_HPB_TEMP 60
#define GUM_PREHEAT_FAN_SPEED 255
/***********************************************************************/

/***********************************************************************
 *************************** UI Language  ******************************
 ***********************************************************************/
#define LANGUAGE_CHOICE 13
/***********************************************************************/
[... 191 lines cut ...]
#define Z_ENDSTOP_SERVO_NR 0
#define DEACTIVATE_SERVOS_AFTER_MOVE
#define Z_ENDSTOP_SERVO_ANGLES {70,155} // Z Axis Extend and Retract angles
/***********************************************************************/

/***********************************************************************
 ****************************** EEPROM *********************************
 ***********************************************************************/
//#define EEPROM_SETTINGS
//#define EEPROM_CHITCHAT
/***********************************************************************/

/***********************************************************************
 ****************************** SDCARD *********************************
 ***********************************************************************/
#define SDSUPPORT
/***********************************************************************/

/***********************************************************************
 ******************************** LCD **********************************
 ***********************************************************************/
#define NEXTION
#define NEXTION_PORT 1
#define NEXTION_GFX
/***********************************************************************/

jgauchia, 2016-03-17:
Hi, 
When extruders are heated up, and printer works correctly when nextion is at splash screen , if you touch the screen later, then works perfectly.

Deepoff, 2016-03-28:
Here's incorrect conclusion the current coordinates on the screen. Disable this code - garbage will disappear

jgauchia, 2016-03-28:
Where is located this code?

Deepoff, 2016-03-28:
yea

jgauchia, 2016-03-28:
Thanks I'll try it

jgauchia, 2016-03-28:
Thanks Deepoff It's works!

Now I'm trying to find why GCODES sent from Nextion doesn't works.

Display something like this.
**
Echo: enqueueing "G91"
Attempted to write invalid text to console, which could be due to an invalid baudrate
Echo: enqueueing "G90"
Attempted to write invalid text to console, which could be due to an invalid baudrate
[ERROR] Traceback (most recent call last):
  File "/home/jordi/Printrun/printrun/printcore.py", line 118, in logError
    try: self.errorcb(error)
  File "/home/jordi/Printrun/printrun/pronsole.py", line 257, in logError
    msg = u"".join(unicode(i) for i in msg)
  File "/home/jordi/Printrun/printrun/pronsole.py", line 257, in <genexpr>
    msg = u"".join(unicode(i) for i in msg)
  File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0x8f in position 25: invalid start byte**

Deepoff, 2016-03-28:
Open new issue for this problem, please.

jgauchia, 2016-03-28:
Ok, I've already open an issue with something similar but with a previous version of MarlinKimbra, 
Also, I'll try Repetier Host,

Deepoff, 2016-03-28:
to send commands in text form you need to use another command . But while this is only my assumption .

jgauchia, 2016-03-28:
Commands are sent touching the different move arrows in Nextion menu

Deepoff, 2016-04-05:
`strcat(buffer, TEST(axis_known_position, Y_AXIS) || !TEST(axis_was_homed, Y_AXIS) ? PSTR(" Y") : PSTR(" ?"));`

`strcat(buffer, TEST(axis_known_position, Z_AXIS) || !TEST(axis_was_homed, Z_AXIS) ? PSTR(" Z") : PSTR(" ?"));`

Here it is necessary to remove code "PSTR" to normal work code watching coordinates.

jgauchia, 2016-04-05:
Ok thank you!

carmelf, 2016-08-08:
Hello , i think i have the same problem 
after connecting the nextion 4.3 to my mini kossel 
the move axis dosent work
and i have the same error on my Arduino terminal : 
"
Echo:enqueueing "G91"
Echo:enqueueing "ep..."
Echo:enqueueing "G90"
Error:Unknown command: "ep..."
"

also i notice that my motor drivers on the ramps 1.4 get extremely hot. 
where should i put this code ?
"
strcat(buffer, TEST(axis_known_position, Y_AXIS) || !TEST(axis_was_homed, Y_AXIS) ? PSTR(" Y") : PSTR(" ?"));

strcat(buffer, TEST(axis_known_position, Z_AXIS) || !TEST(axis_was_homed, Z_AXIS) ? PSTR(" Z") : PSTR(" ?"));" 

maybe this is my problem ?

jgauchia, 2016-08-09:
If motor drivers are extremely hot , try to adjust voltage level of the stepper drivers (pololus)
I think that this "piece" of code it's correct in last version of firmware, it's inside Nextion_LCD.cpp 

This isn't the problem , i think, I didn't solve it...