MKFirmware/MarlinKimbra #108: error : sd card init

prophetmaster, 2016-03-28:
message to serial on connect : 

I have ramps 1.4 and in Configuration_Feature.h

#define SDSUPPORT
#define SD_DISABLED_DETECT
//#define SD_DETECT_INVERTED
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

on connect i get this error: SD card init fail...

MagoKimbra, 2016-03-28:
If enabled SD DETECT DISABLED firmware no longer controls the insertion pin SD and assumes that it is always on, so when he goes to read and can not find it by mistake.

prophetmaster, 2016-03-28:
I tested and not it does not work!

MagoKimbra, 2016-03-28:
Excuse me, I explained wrong ... it is normal that if it do not find sd give error. Because for the firmware the SD is always inside.

prophetmaster, 2016-03-28:
this makes little sense as a way to ... well how do we make it work?

MagoKimbra, 2016-03-28:
Some sd readers do not have the pin to check if the sd is inserted or not.
So you have to force the firmware to believe that it is always in, but if it is not inserted give error.
If your sd reader has the pin disable SD_DISABLED_DETECT

prophetmaster, 2016-03-28:
I tested all possible solutions nothing works ...

MagoKimbra, 2016-03-28:
Is Sd formatted FAT? The reader function?

prophetmaster, 2016-03-28:
my sdcard is dead i have tested another and no bug... excuse me for the issue

MagoKimbra, 2016-03-28:
No problem...

closed 2016-03-28 by MagoKimbra