MKFirmware/MarlinKimbra #110: autoleveling G29

prophetmaster, 2016-03-28:
Send: G29
Recv: Echo: 0.240 0.210 -0.023 -0.130 -0.237 -0.320 -0.370 -0.300 0.790
Recv: Echo: -0.037 0.100 0.077 0.440 0.397 0.297 0.237 0.513 0.667
Recv: Echo: -0.153 -0.097 0.383 -0.033 -0.400 -0.707 -0.977 -1.387 -0.010
Recv: Echo: -0.063 0.293 0.207 0.393 0.570 0.433 0.070 -0.557 0.213
Recv: Echo: -0.230 0.003 -0.247 -0.483 -0.780 -0.920 -0.793 -0.157 0.437
Recv: Echo: -0.710 1.800 1.887 1.847 1.640 1.257 0.740 0.167 0.500
Recv: Echo: -1.190 0.453 -0.890 -2.133 -3.190 -3.960 -4.350 -4.497 0.563
Recv: Echo: 0.543 0.457 3.430 3.493 2.650 0.907 -1.567 4.550 4.940
Recv: Echo: 0.397 -6.517 -7.047 -6.423 -5.773 -4.077 -2.380 -0.683 6.660
Recv: X:0.00 Y:0.00 Z:20.00 E:0.00 Count Alpha:25285 Beta:25285 Teta:25285

no comment ! -6.660mm ??? with my mecanical comparator +-0.1mm default...

MagoKimbra, 2016-03-28:
Sorry i do not understand!!!

prophetmaster, 2016-03-28:
well it's just not possible that measures more than 1 mm of default on the bed

MagoKimbra, 2016-03-28:
why not?

prophetmaster, 2016-03-28:
because if I mechanically measured with a probe I find maximum 0.1mm fake on the bed! So how come you found it 6mm?

MagoKimbra, 2016-03-28:
Respect to center, first probe is center this is zero...

prophetmaster, 2016-03-28:
I did not understand but I want to know why it does not work?

MagoKimbra, 2016-03-28:
ok write the G30 X0 Y0 command

prophetmaster, 2016-03-28:
do I send you my config and you tell me if I'm wrong somewhere?

MagoKimbra, 2016-03-28:
The firmware is a first probe to a center, and calculates the difference between the point the probe and the height. If it's ok should give 0.
After various probes, and it calculates the difference between that point and the center. If the floor is not level by those values.

prophetmaster, 2016-03-28:
I get this : Echo: Bed Z-Height at X:0.00 Y:0.00 = -1.0933

prophetmaster, 2016-03-28:
and my config :  

Send: M501
Recv: Echo: V25 stored settings retrieved (335 bytes)
Recv: Config: Steps per unit:
Recv: Config:   M92 X100.00 Y100.00 Z100.00 E99.35
Recv: Config: Maximum feedrates (mm/s):
Recv: Config:   M203 X500.00 Y500.00 Z500.00 E100.00
Recv: Config: Maximum Acceleration (mm/s2):
Recv: Config:   M201 X5000 Y5000 Z5000 E1000
Recv: 
Recv: Config: Accelerations: P=printing, V=travel and T* R=retract
Recv: Config:   M204 P3000.00 V3000.00
Recv: Config:   M204 T0 R10000.00
Recv: Config: Advanced variables: S=Min feedrate (mm/s), V=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: Config:   M205 S0.00 V0.00 B20000 X20.00 Z20.00 E5.00
Recv: Config: Home offset (mm):
Recv: Config:   M206 X0.00 Y0.00 Z0.00
Recv: Config: Delta Geometry adjustment:
Recv: Config:   M666 A0.000 B0.000 C0.000 I0.000 J0.000 K0.000 U0.000 V0.000 W0.000 R178.75 D291.55 H191.50
Recv: Config: Endstop Offsets:
Recv: Config:   M666 X0.00 Y0.00 Z0.00
Recv: Config: Z-Probe Offset:
Recv: Config:   M666 P X0.00 Y20.00 Z-1.00
Recv: Config: PID settings:
Recv: Config:   M301 H

MagoKimbra, 2016-03-28:
OK your High is correctly? Down width Z and check is nozzle touch the plain at zero...

prophetmaster, 2016-03-28:
No there is 1 mm

MagoKimbra, 2016-03-28:
Ok Increment high +1 width m666 h command

prophetmaster, 2016-03-28:
Recv: Echo: Bed Z-Height at X:0.00 Y:0.00 = -0.0467

MagoKimbra, 2016-03-28:
It's ok, now the 0.046 is possible error for Z PROBE OFFSET...
Now when you give command G29 the firmware chek the center this point is zero.. When check another poin the value is differenze from center point... If you have big difference you must do the autocalibration width G30 A command...

prophetmaster, 2016-03-28:
Ok i send a G29...

prophetmaster, 2016-03-28:
Recv: Echo: 1.433 0.887 0.297 0.207 0.323 0.467 0.610 0.547 -0.243
Recv: Echo: -0.810 0.813 0.477 0.137 0.150 0.163 0.203 -0.247 -0.697
Recv: Echo: -0.777 -0.647 0.193 0.067 -0.023 -0.140 -0.250 -0.280 -0.900
Recv: Echo: -0.483 -0.483 -0.483 -0.427 -0.343 -0.253 -0.203 -0.153 -0.180
Recv: Echo: -0.450 -0.320 -0.190 -0.030 0.137 0.347 0.593 0.830 0.993
Recv: Echo: 2.020 1.627 1.233 0.863 0.577 0.377 0.353 0.427 0.670
Recv: Echo: 3.443 1.123 0.627 0.243 -0.110 -0.460 -0.870 -1.433 -2.190
Recv: Echo: 4.867 0.390 -4.250 -3.117 -2.360 -1.930 -1.830 -1.910 -3.220
Recv: Echo: 0.153 -8.743 -6.123 -2.917 -1.713 -0.610 0.577 -2.387 -4.250

I SEND A G30 A...

prophetmaster, 2016-03-28:
Send: G29
Recv: Echo: -0.370 -0.313 -0.180 -0.100 -0.133 -0.203 -0.233 -0.177 0.333
Recv: Echo: 0.373 -0.203 -0.187 -0.080 -0.057 -0.063 -0.050 0.233 0.360
Recv: Echo: 0.350 0.247 -0.037 -0.060 0.020 0.077 0.133 0.163 0.257
Recv: Echo: 0.227 0.173 0.120 0.130 0.067 0.033 0.023 0.067 0.153
Recv: Echo: 0.130 0.063 -0.003 -0.007 -0.033 -0.060 -0.123 -0.193 -0.273
Recv: Echo: -0.650 -0.527 -0.403 -0.310 -0.223 -0.183 -0.167 -0.203 -0.233
Recv: Echo: -1.050 -0.320 -0.120 0.047 0.180 0.307 0.430 0.630 0.850
Recv: Echo: -1.120 0.070 1.260 0.913 0.700 0.623 0.590 0.697 1.427
Recv: Echo: 0.260 2.473 1.647 0.773 0.353 -0.073 -0.563 0.763 2.003
Recv: X:0.00 Y:0.00 Z:20.00 E:0.00 Count Alpha:25033 Beta:25158 Teta:25033

MagoKimbra, 2016-03-28:
The autocalibration may take time. Can be up to one hour ...
I'm going to sleep... Tomorrow i read you result..

prophetmaster, 2016-03-28:
Ok me too, thanks for the help and good night !

MagoKimbra, 2016-03-28:
This is center line probe
Recv: Echo: 0.130 0.063 -0.003 -0.007 -0.033 -0.060 -0.123 -0.193 -0.273
Look the center point is ~0 but the extreme point front and behind no.. You have problem width delta radius... The autocalibration correct it..

prophetmaster, 2016-03-30:
ok after some test G30 A G29 my sensor is not best for me !