MKFirmware/MarlinKimbra #112: Servo not retract when the printer is turned on

RicardoGA, 2016-03-30:
hi when i turn on the printer the servo won't retract or go to rest position, in previous versions (4.1.2) when the printer is turned on the servo is activated then it's retracted and then deactivated to avoid jitter.

now in the version 4.2.8 when the printer is turned on the servo is activated and then deactivated, 

i let the link of my configuration files
thanks in advance

[https://www.dropbox.com/sh/2zkh17jjtb7jhy3/AADDypup4UBnO-jaIkLnYFtha?dl=0](url)

jgauchia, 2016-03-30:
Can't view the file.
In my printer, with 4.2.8 firmware the servo works well.

RicardoGA, 2016-03-30:
i just check the link and it work but you have to copy the addres i don´t know why the direct link don´t work

i leave another link just in case

https://onedrive.live.com/?authkey=!AKNkSc35X1DRzGA&id=E1DAE8C25D907900!1585&cid=E1DAE8C25D907900

MagoKimbra, 2016-03-30:
When init servo i write the retract angle. You put {110,40}, is 40 retract angle?

jgauchia, 2016-03-30:
In my firmware was: #define Z_ENDSTOP_SERVO_ANGLES {70,155}

RicardoGA, 2016-03-30:
@MagoKimbra that's rigth my retraction angle is 40

RicardoGA, 2016-03-30:
@jgauchia @MagoKimbra right now i'm printing but when it finish, i'll make a video of the problem

jgauchia, 2016-03-30:
@RicardoGA ok, but I'll see tomorrow, here is too late :P

RicardoGA, 2016-03-30:
here's the link of the video

https://www.dropbox.com/s/wfcabmk5pa1ov4y/Video%2030-03-16%204%2058%2034%20p.m..mov?dl=0

jgauchia, 2016-03-31:
Hi , I saw in your video servo moves when power up the power supply?

Are you send a GCODE to move servo, like M280 to set angle of servo?
Wiring it's correct? (servo and ramps)

RicardoGA, 2016-03-31:
@jgauchia The wiring is correct and the servo moves with the command M280 and also when i home the axis (G28) and whe use bed leveling (G29) works great but when the printer is turned on he servo don't go to rest position (retraction) and in previois version of marlin kimbra when the printer is turned on th servo move to rest position

jgauchia, 2016-03-31:
@RicardoGA mmm ok I see