MKFirmware/MarlinKimbra #116: Dual X-carriage enable problem

dp90pusika, 2016-04-11:
When activating dual x carriage, sanitycheck wont let it compile:
#error DEPENDENCY ERROR: Missing or invalid definitions for DUAL_X_CARRIAGE mode.
Pins are sorted out, steppers are defined, but it wont let me compile.

jgauchia, 2016-05-28:
The same error:

#error DEPENDENCY ERROR: Missing or invalid definitions for DUAL_X_CARRIAGE mode.

jgauchia, 2016-05-28:
Hi, first of all , sorry for my english

In pins.h

/****************************************************************************************/
#include "../Configuration_Pins.h"
/****************************************************************************************/

#if X_HOME_DIR > 0    // Home X to MAX
  #undef X_MIN_PIN
  #define X_MIN_PIN -1
#elif X_HOME_DIR < 0  // Home X to MIN
  #undef X_MAX_PIN
  #define X_MAX_PIN -1
#endif //X_HOME_DIR > 0

When it's defined DUAL_X_CARRIAGE, X_HOME_DIR was -1 and X2_HOME_DIR was 1 .
Well if X_HOME_DIR was -1 then X_MAX_PIN was undefined and sanity checks raising the error.
Maybe in pins.h checking if DUAL_X_CARRIAGE was defined X_MAX_PIN doesn't be undefined???

I'll try comment this code , and error doesn't appears but gets another compiling error:

 _sketch/module/motion/stepper.cpp: In function 'void update_endstops()':
sketch/module/motion/stepper.cpp:382:31: error: 'struct block_t' has no member named 'active_extruder'
           if ((current_block->active_extruder == 0 && X_HOME_DIR == -1) || (current_block->active_extruder != 0 && X2_HOME_DIR == -1))
                               ^
sketch/module/motion/stepper.cpp:382:92: error: 'struct block_t' has no member named 'active_extruder'
           if ((current_block->active_extruder == 0 && X_HOME_DIR == -1) || (current_block->active_extruder != 0 && X2_HOME_DIR == -1))
                                                                                            ^
sketch/module/motion/stepper.cpp:393:31: error: 'struct block_t' has no member named 'active_extruder'
           if ((current_block->active_extruder == 0 && X_HOME_DIR == 1) || (current_block->active_extruder != 0 && X2_HOME_DIR == 1))
                               ^
sketch/module/motion/stepper.cpp:393:91: error: 'struct block_t' has no member named 'active_extruder'
           if ((current_block->active_extruder == 0 && X_HOME_DIR == 1) || (current_block->active_extruder != 0 && X2_HOME_DIR == 1))
                                                                                           ^
exit status 1_

I've a RAMPS board with two extruders, I'll add a extra stepper board controller for dual x carriage.

jgauchia, 2016-06-04:
With new version and changed X_MIN_PIN and X_MAX_PIN something like (to test it):

_#if DISABLED(DUAL_X_CARRIAGE)
	#if X_HOME_DIR > 0    // Home X to MAX
		#undef X_MIN_PIN
		#define X_MIN_PIN -1
	#elif X_HOME_DIR < 0  // Home X to MIN
		#undef X_MAX_PIN
		#define X_MAX_PIN -1
	#endif //X_HOME_DIR > 0
#endif._

new compiling error raised:

_In file included from sketch/module/../base.h:4:0,
                 from sketch/module/MK_Main.cpp:31:
sketch/module/MK_Main.cpp: In function 'void update_software_endstops(AxisEnum)':
sketch/module/MK_Main.cpp:1171:30: error: 'extruder_offset' was not declared in this scope
       float dual_max_x = max(extruder_offset[X_AXIS][1], X2_MAX_POS);
                              ^
/home/jordi/arduino-1.6.8/hardware/arduino/avr/cores/arduino/Arduino.h:85:20: note: in definition of macro 'max'
 #define max(a,b) ((a)>(b)?(a):(b))
                    ^
sketch/module/MK_Main.cpp:1179:73: error: 'duplicate_extruder_x_offset' was not declared in this scope
         sw_endstop_max[X_AXIS] = min(base_max_pos(X_AXIS), dual_max_x - duplicate_extruder_x_offset) + offs;
                                                                         ^
/home/jordi/arduino-1.6.8/hardware/arduino/avr/cores/arduino/Arduino.h:84:24: note: in definition of macro 'min'
 #define min(a,b) ((a)<(b)?(a):(b))
                        ^
exit status 1_

jgauchia, 2016-06-04:
@MagoKimbra seems to definition of extruder_offset like original marlin wasn't created.

_// Extruder offsets
#if EXTRUDERS > 1
  #ifndef EXTRUDER_OFFSET_X
    #define EXTRUDER_OFFSET_X { 0 } // X offsets for each extruder
  #endif
  #ifndef EXTRUDER_OFFSET_Y
    #define EXTRUDER_OFFSET_Y { 0 } // Y offsets for each extruder
  #endif
  float extruder_offset[][EXTRUDERS] = {
    EXTRUDER_OFFSET_X,
    EXTRUDER_OFFSET_Y
    #if ENABLED(DUAL_X_CARRIAGE)
      , { 0 } // Z offsets for each extruder
    #endif
  };
#endif_

and in MK_main.cpp

_        }
       else if (dual_x_carriage_mode == DXC_DUPLICATION_MODE) {
         sw_endstop_min[X_AXIS] = base_min_pos(X_AXIS) + offs;
-        sw_endstop_max[X_AXIS] = min(base_max_pos(X_AXIS), dual_max_x - duplicate_extruder_x_offset) + offs;
+        sw_endstop_max[X_AXIS] = min(base_max_pos(X_AXIS), dual_max_x - duplicate_hotend_x_offset) + offs;
         return;
       }
     }_

can you test if it's correct?

jgauchia, 2016-06-08:
Well, really doesn't works ... With Marlin and dual X , no problems... :(

jgauchia, 2016-08-16:
Dual X carriage works¿?

dp90pusika, 2016-08-16:
# I tested it recently, nothing is changed, it doesn't work at all
With dual x carriage enabled I get this sanitycheck.h compiling error:

In file included from sketch\base.h:52:0,

                 from C:\Users\Dani\Desktop\Marlin 32_08.08\MarlinKimbra4due-master\MK4due\MK4due.ino:216:

sketch\src/sanitycheck.h:1669:8: error: #error DEPENDENCY ERROR: Missing or invalid definitions for DUAL_X_CARRIAGE mode.

       #error DEPENDENCY ERROR: Missing or invalid definitions for DUAL_X_CARRIAGE mode.

        ^

exit status 1
Error compiling for board Arduino Due (Programming Port).

# After commenting out the lines in sanitycheck.h:
 /**
   * Dual X Carriage requirements
   */
  /*#if ENABLED(DUAL_X_CARRIAGE)
    #if EXTRUDERS == 1 || MECH(COREXY) \
        || HASNT(X2_ENABLE) || HASNT(X2_STEP) || HASNT(X2_DIR) \
        || DISABLED(X2_HOME_POS) || DISABLED(X2_MIN_POS) || DISABLED(X2_MAX_POS) \
        || HASNT(X_MAX)
      #error DEPENDENCY ERROR: Missing or invalid definitions for DUAL_X_CARRIAGE mode.
    #endif
    #if X_HOME_DIR != -1 || X2_HOME_DIR != 1
      #error CONFLICT ERROR: Please use canonical x-carriage assignment.
    #endif
  #endif // DUAL_X_CARRIAGE
*/

# Causes this compiling error:

sketch\src\MK_Main.cpp: In function 'void tool_change(uint8_t, float, bool)':

sketch\src\MK_Main.cpp:8667:78: error: 'DEBUG__INFO_POS' was not declared in this scope

         if (DEBUGGING(INFO)) DEBUG__INFO_POS("New Extruder", current_position);

                                                                              ^

In file included from sketch\src\../base.h:65:0,

                 from sketch\src\MK_Main.cpp:31:

sketch\src\MK_Main.cpp:8671:59: error: 'inactive_extruder_x_pos' was not declared in this scope

             current_position[X_AXIS] = LOGICAL_X_POSITION(inactive_extruder_x_pos);

                                                           ^

sketch\src\../src/MK_Main.h:122:38: note: in definition of macro 'LOGICAL_POSITION'

 #define LOGICAL_POSITION(POS, AXIS) (POS + home_offset[AXIS] + position_shift[AXIS])

                                      ^

sketch\src\MK_Main.cpp:8671:40: note: in expansion of macro 'LOGICAL_X_POSITION'

             current_position[X_AXIS] = LOGICAL_X_POSITION(inactive_extruder_x_pos);

                                        ^

sketch\src\MK_Main.cpp:8675:13: error: 'active_extruder_parked' was not declared in this scope

             active_extruder_parked = (active_extruder == 0); // this triggers the second extruder to move into the duplication position

             ^

sketch\src\MK_Main.cpp:8679:64: error: 'duplicate_extruder_x_offset' was not declared in this scope

               current_position[X_AXIS] = destination[X_AXIS] + duplicate_extruder_x_offset;

                                                                ^

sketch\src\MK_Main.cpp:8681:13: error: 'extruder_duplication_enabled' was not declared in this scope

             extruder_duplication_enabled = false;

             ^

In file included from sketch\src\../base.h:54:0,

                 from sketch\src\MK_Main.cpp:31:

sketch\src\MK_Main.cpp:8693:54: error: 'active_extruder_parked' was not declared in this scope

           ECHO_LMT(INFO, "Active extruder parked: ", active_extruder_parked ? "yes" : "no");

                                                      ^

sketch\src\../src/communication/communication.h:45:54: note: in definition of macro 'SERIAL_PRINT'

   #define SERIAL_PRINT(msg, args...)  MKSERIAL.print(msg, ##args)

                                                      ^

sketch\src\../src/communication/communication.h:91:57: note: in expansion of macro 'ECHO_MT'

   #define ECHO_LMT(srt, msg, txt)           ECHO_S(srt),ECHO_MT(msg, txt),ECHO_E

                                                         ^

sketch\src\MK_Main.cpp:8693:11: note: in expansion of macro 'ECHO_LMT'

           ECHO_LMT(INFO, "Active extruder parked: ", active_extruder_parked ? "yes" : "no");

           ^

exit status 1
Error compiling for board Arduino Due (Programming Port).

dp90pusika, 2016-08-16:
right now I use this branch of marlin wich works for me: https://github.com/Wurstnase/Marlin4Due

And I'm testing https://github.com/esenapaj/Marlin for DRV8825 drivers because its broken right now, but everithing els is working, if you have diferent stepper driwers it could be working for you

jgauchia, 2016-08-16:
I'm using DRV8825 steppers for axis movements and Ardunio Mega , not DUE.
I need to work Dualcarriage and Nextion LCD,

dp90pusika, 2016-08-16:
I have a working version for the Mega whit DRV8825 and dual x carriage calibrated, but I don't know anithing about setting up that LCD. If you give me an email addres I share it with you, saddly I don't remember which branch or version was it. So I have this one copy of it

jgauchia, 2016-08-16:
It's a branch of MarlinKimbra?

dp90pusika, 2016-08-16:
no saddly, it's from the main branch

jgauchia, 2016-08-16:
Ok, no problem, I'm using Marlin main branch now, with standard LCD screen.
Works well.

dp90pusika, 2016-08-16:
Ok, than we use about the same firmware right now. 
I hope the developer will give you a hand with Kimbra soon.

MagoKimbra, 2016-08-16:
Ok i check this... Your board?

dp90pusika, 2016-08-16:
@jgauchia has a RAMPS 1.4 with arduino mega and an expansion board for one extra stepper.

I use a RADDS board with an arduino due

MagoKimbra, 2016-08-16:
Ramps with expansion board, but the pins are mapped ?? Because the ramps is not pins to X2 !!!

dp90pusika, 2016-08-16:
Sorry I don't really have the authority to speak in the name of jaguar, he is probably sleeping right now, we are about in the same time zone and it's midnight here.
But I follow him on instagram and is seen her machine working correctli with the old Marlin main branch. So It is probbabli mapped correctly. 
And allso when I was testing MarlinKimbra I definiteli mapped the step, dir, enable pin s for the second x axes and still got this compiling error.

MagoKimbra, 2016-08-16:
ok... Thanks for info... Check this..

dp90pusika, 2016-08-16:
quick update:
I was double checking Configuration_Pins.h and found this:
#if ENABLED(X2_IS_TMC)
  #define X2_ENABLE_PIN 49
  #define X2_STEP_PIN   51
  #define X2_DIR_PIN    53
#endif
but this has to be :
#if ENABLED(DUAL_X_CARRIAGE)
  #define X2_ENABLE_PIN 49
  #define X2_STEP_PIN   51
  #define X2_DIR_PIN    53
#endif

Now it registers my pi setting and mowes on to the second compiling error:
[Kimbra compiling error.txt](https://github.com/MagoKimbra/MarlinKimbra/files/421437/Kimbra.compiling.error.txt)

jgauchia, 2016-08-17:
Hi,
Pins are mapped to 43,45 and 47 ,
Arduino Mega with a RAMPS 1.4 + Extra stepper
Nextion LCD

jgauchia, 2016-08-17:
Now I'm upgrading bed for my printer, If you need to test the firmware I can do it without problems.

jgauchia, 2016-09-09:
Hi! 
With new version MK4Duo seems to works well!

dp90pusika, 2016-09-09:
for Due it's still not good i got this error:

In file included from sketch\base.h:55:0,

                 from C:\Users\Dani\Desktop\Marlin 32_08.08\09.09\MarlinKimbra-4_2_9\MK\MK.ino:216:

sketch\src/HAL/HAL.h:57:22: fatal error: avr/io.h: No such file or directory

   #include <avr/io.h>

There is no io.h file in any of the files, if I comment out the missing includes, then it can't find the pinout of the arduino due.

jgauchia, 2016-09-09:
@dp90pusika ammm ok, sorry
For Arduino Mega it's good.

dp90pusika, 2016-09-09:
@jgauchia no problem, just tested it and gave some feedback

bruce356, 2016-10-13:
@MagoKimbra, Hi I also have the Due + RADDS board with a Mendelmax Cartesian printer, drivers are DRV8825. I have installed dual X-Carriage using separate motors and drivers (using E3 driver for X2, Pins used are - enable-49, step-51, dir-53).

How to configure MK4DUO to make this work?

Whenever I enable //#define DUAL_X_CARRIAGE (in Configuration_Feature.h) on compiling (Arduino v1.6.11) this error pops up - LOGICAL_X_POSITION(inactive_extruder_x_pos).

Have you had a chance to look at this, or is my config method wrong and if so how to fix it please.

Thanks for any help - Regards