MKFirmware/MarlinKimbra #119: Wrong output on M105

repetier, 2016-04-23:
Your output to M105 is wrong. It looks like this:

< 17:44:02.233: N4054 M105
> 17:44:02.245: ok T:21.1 /0.0 B:23.0 /0.0 T :21.1 /0.0 T:22.8 /0.0 B@:0 @:0 @ :0 @:0

The second T: has a 0 byte between T and : where you wanted to add a "0" char. Same with @1: which is written as "@"+chr(1)+":". So you always forget to add the offset for zero char. That causes problems for parsers. In C it would even cut the string at the first chr(0). This prevents host from correctly parsing the temperatures.

MagoKimbra, 2016-04-23:
Thank's
I see to fix it!!!

repetier, 2016-04-27:
Looks much better now. Just a tine thing I noted that you output

okT

should be
ok T

MagoKimbra, 2016-04-27:
Yes i know... This night i correct It...

MagoKimbra, 2016-04-27:
Fix!!

closed 2016-04-27 by MagoKimbra