MKFirmware/MarlinKimbra #122: Wait is wait

repetier, 2016-04-26:
One litte more violation of protocols. You implemented "wait" to signal queue is empty so we get it every second. But you send "Wait" with capital w, which is not recogniced as all use "wait". Having different solutions only makes hosts only slower and older would never understand.

repetier, 2016-04-26:
Speaking about waits I have the feeling you are sending it sometime at wrong time, see this log with added comments.

<  6:43:09.909: N8653 G1 X345.59 Y210.42 E15.8037
>  6:43:09.909: ok N8652 P0 B2

**Move here is 50,4mm at 55mm/s so with acceleration maybe just > 1s**

<  6:43:09.909: N8654 G1 X309.97 Y174.8 E16.4354
>  6:43:10.012: ok N8654 P0 B1

<  6:43:10.013: N8655 G1 X309.28 Y174.36 E16.4456
>  6:43:10.013: ok N8654 P0 B2
<  6:43:10.013: N8656 G1 X295.26 Y168.12 E16.638
>  6:43:10.116: ok N8656 P0 B1
<  6:43:10.116: N8657 G1 X277.55 Y154.51 E16.9181
>  6:43:10.117: ok N8656 P0 B2
<  6:43:10.117: N8658 G1 X276.36 Y153.49 E16.9378
>  6:43:10.117: ok N8656 P0 B3
<  6:43:10.117: N8659 G1 X274.99 Y151.98 E16.9634
>  6:43:10.220: ok N8659 P0 B1
<  6:43:10.220: N8660 G1 X273.83 Y150.25 E16.9895
>  6:43:10.221: ok N8659 P0 B2
<  6:43:10.221: N8661 G1 X272.89 Y148.23 E17.0174
>  6:43:10.221: ok N8659 P0 B3
<  6:43:10.222: N8662 G1 X272.28 Y146.18 E17.0443
>  6:43:10.324: ok N8662 P0 B1
<  6:43:10.325: N8663 G1 X271.99 Y144.04 E17.0713
>  6:43:10.325: ok N8662 P0 B2
<  6:43:10.325: N8664 G1 X272.02 Y141.84 E17.0989
>  6:43:10.428: ok N8662 P0 B3
<  6:43:10.428: N8665 G1 X272.36 Y139.82 E17.1246
>  6:43:10.428: ok N8664 P0 B2
<  6:43:10.428: N8666 G1 X272.99 Y137.82 E17.1509
>  6:43:10.429: ok N8664 P0 B3
<  6:43:10.429: N8667 G1 X273.96 Y135.88 E17.1781
>  6:43:10.532: ok N8667 P0 B1
<  6:43:10.532: N8668 M105
<  6:43:10.533: N8669 G1 X274.8 Y134.61 E17.1971
>  6:43:10.533: ok N8667 P0 B2
>  6:43:10.533: ok N8667 P0 B3
<  6:43:10.533: N8670 G1 X275.68 Y133.54 E17.2145
>  6:43:10.637: ok T:255.0 /255.0 B:120.4 /120.0 T:255.0 /255.0 T:79.8 /0.0 T:78.8 /0.0 T:80.4 /80.0 B@:0 @:46 @:46 @:0 @:0 @:102
<  6:43:10.637: N8671 G1 X276.49 Y132.73 E17.229
>  6:43:10.637: ok N8670 P0 B2
<  6:43:10.638: N8672 G1 X276.32 Y132.57 E17.229 F15000
>  6:43:11.046: ok N8672 P0 B1
<  6:43:11.047: N8673 G1 X276.26 Y132.33 E17.232 F1800
>  6:43:11.252: ok N8673 P0 B1
<  6:43:11.252: N8674 G1 X277.56 Y131.26 E17.253
>  6:43:11.253: ok N8673 P0 B2
<  6:43:11.253: N8675 G1 X278.84 Y130.42 E17.2722

**1s delay from above long move, that is no wait reason, input queue is not empty!**

>  6:43:12.274: ok N8675 P0 B1
<  6:43:12.274: N8676 G1 X280.75 Y129.47 E17.299
>  6:43:12.275: ok N8675 P0 B2
<  6:43:12.275: N8677 G1 X282.7 Y128.84 E17.3247
>  6:43:12.276: ok N8675 P0 B3
<  6:43:12.276: N8678 G1 X284.32 Y128.54 E17.3453

**Wait from 1s above I guess**

>  6:43:12.276: Wait
>  6:43:12.379: ok N8676 P2 B3
<  6:43:12.379: N8679 G1 X286.53 Y128.39 E17.3731
>  6:43:12.379: ok N8678 P1 B2

I also think you send wait if a action takes longer then 1s and not if your input queue is empty for > 1s.

Together with octoprint/Marlin we have also a new protocol for long operations, so hosts know that the firmware is just busy and do not timeout connection.See
https://github.com/MarlinFirmware/Marlin/issues/3068 for more informations how that works. These are different things which need different handling on host side and using them wrong can cause new communication errors.

MagoKimbra, 2016-04-27:
I think have fix this!!!

MagoKimbra, 2016-05-05:
Please i have I need help...
I can not operate ADVANCE extruder in MK4due ...
I do not know well timed functions ... Can you give me a hand in creating the right HAL timer?
Thank's!!!

repetier, 2016-05-06:
Isn't it coupled to move steps?

What we do and what iv'e seen in marlin is a extruder stepper interrupt that is simply faster then normally needed so it executes all added extra steps in a timely fashion. I think it is extremely hard to change extruder timer speed from bresenham timer to do it more correct. If you do it is the same as for bresenham interrupt or however marlin calls the stepper interrupt.

closed 2016-05-06 by repetier