MKFirmware/MarlinKimbra #130: error auto calibration in delta ( G30 A)

Briangsouza, 2016-05-07:
the movement between the probe this very concave up, something is strange

MagoKimbra, 2016-05-08:
for now it is normal. During the calibration phase must make movements that take no account of what is calibrating it will not work ... I'm studying another solution. Put raise between z very high from 20 probe up even 30 ...

karabas, 2016-05-16:
Briangsouza, try to set deltaradius as close to real as possible before calibration.

MagoKimbra, look at smoothieware deltacalibration forks LSQ and pilot626

lsgarrow, 2016-06-28:
Another strange issue, but not sure if related:

After probing the bed the 1st or 2nd sequence, probe returns to center and repeats there as if stuck sequence, like broken record. LCD is still responsive, showing Autocalibration is occurring, but unable to stop unless reset controller or hit estop on Rep. host. 

I have tried clear eeprom/re-upload, delta geometry is close to real as possible to nearest mm.
Using RAMPS 1.4 and never had issue with MK before.

lsgarrow, 2016-06-30:
I loaded an old version and it recognized that my end stops were not within 3mm of each other. I adjusted until the G30 A would work for the old firmware. After installing latest firmware, I ran G30 D to check diag. rods, and after that G30 A worked. The center point still repeated the up and down motion for more than 5 times but ended up moving onto the next point. Hope this helps.