MKFirmware/MarlinKimbra #134: Does not compile for delta

karabas, 2016-05-16:
MK 4.2.8; arduino 1.6.9; RRD Smart controller; 2560 + ramps14
sketch\module\lcd\ultralcd.cpp.o: In function `line_to_current(AxisEnum)':
sketch\module\lcd/ultralcd.cpp:511: undefined reference to `calculate_delta(float*)'
collect2.exe: error: ld returned 1 exit status
exit status 1

MagoKimbra, 2016-05-16:
I try and have not error...

karabas, 2016-05-16:
I removed "static" from void declaration and now it compiles. What IDE use use?
Arduino IDE does not see folders for editing.

fovezea, 2016-06-20:
Hi guys, I have the same problem. Just downloaded the repository and change to "delta". Before change to "delta" was compiling. I removed the "static" from the declaration but still does not work.

Error message as below;
Arduino: 1.6.9 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
sketch\module\MK_Main.cpp.o: In function `prepare_delta_move_to(float*)':
sketch\module/MK_Main.cpp:9574: undefined reference to `adjust_delta(float*)'
collect2.exe: error: ld returned 1 exit status
exit status 1

closed 2016-06-20 by MagoKimbra

fovezea, 2016-06-20:
Thanks MagoKimbra! I can confirm that is working. It compiles now without problem.

commit e3e15a66 (2016-06-20): Fix #134