MKFirmware/MarlinKimbra #141: MK4DUE not compiled.

pirat777knj, 2016-05-30:
Greetings. All good!
I Bought Arduino Duje. But Marlin Kimbra firmware compiles a bunch of errors. That is not compiled at all. How do I fix this?

[Uploading errors.txt…]()

MagoKimbra, 2016-05-30:
The link error is broke...

Date: Mon, 30 May 2016 13:32:33 -0700

pirat777knj, 2016-05-31:
[errors.txt](https://github.com/MagoKimbra/MarlinKimbra/files/291137/errors.txt)

Here is the complete error log .

MagoKimbra, 2016-05-31:
Ok you must setting ide warning to none. Go in setting Arduino ide and setting warning to none....
Do you what display set??

pirat777knj, 2016-05-31:
Yes. Reprap fullgraphic smart controller + adapter 5V-3.3V.

MagoKimbra, 2016-05-31:
The Display Fullgraphic is not supported for all cards Arduino due.
Have you card?

pirat777knj, 2016-06-01:
of course I have. but even with the flashing turn off the display gives an error (

pirat777knj, 2016-06-03:
You say that the display is not supported. But on firmware Repetier Firmware it works. True adapter.
[image]

MagoKimbra, 2016-06-03:
No I meant in MK4due, I have not the right pin for the display of all the cards.
If you have pin write me and i add this in firmware!!!

pirat777knj, 2016-06-04:
If you need the pin numbers on the board Due corresponding conclusions of the display, here:
according to the scheme:
[http://forums.reprap.org/file.php?249,file=49666,filename=RRD_FULL_GRAPHIC_SMART_CONTROLER_SCHEMATIC.pdf,download=1](url)
As well as the scheme:
[http://www.reprap.org/mediawiki/images/3/3f/Arduinomegapololushieldschematic.png](url)

>LCD RW pin 16
>LCD RS pin 17
>LCD E pin 23

Checked multimeter. All correspondence. Also, I decided to write conclusions for thermistors.

> #define TEMP_0_PIN         9   
> #define TEMP_1_PIN         10 

Download the latest update, for some reason, was to swear:

> In file included from sketch\base.h:51:0,

                 from C:\Users\tj\Desktop\MK4due\MK4due.ino:210:

sketch\module/sanitycheck.h:592:8: error: #error DEPENDENCY ERROR: Missing setting Z_RAISE_BEFORE_HOMING

       #error DEPENDENCY ERROR: Missing setting Z_RAISE_BEFORE_HOMING

        ^

exit status 1
Ошибка компиляции для платы Arduino Due (Programming Port).

closed 2016-06-17 by MagoKimbra