MKFirmware/MarlinKimbra #143: AUTO BED LEVELING ISSUE

jgauchia, 2016-06-04:
If AUTO_BED_LEVELING_FEATURE was enabled and MIN_Z_HEIGHT_FOR_HOMING wasn't enabled sanitycheck raise this error:

_sketch/module/sanitycheck.h:1954:6: error: #error "Z_RAISE_BEFORE_HOMING is deprecated. Use MIN_Z_HEIGHT_FOR_HOMING instead."
     #error "Z_RAISE_BEFORE_HOMING is deprecated. Use MIN_Z_HEIGHT_FOR_HOMING instead."_

When MIN_Z_HEIGHT_FOR_HOING was defined and Z_RAISE_BEFORE_HOMING was undefined sanitycheck raise this error:

_sketch/module/sanitycheck.h:592:8: error: #error DEPENDENCY ERROR: Missing setting Z_RAISE_BEFORE_HOMING
       #error DEPENDENCY ERROR: Missing setting Z_RAISE_BEFORE_HOMING_

this has no logic

MagoKimbra, 2016-06-04:
Fix it!

closed 2016-06-04 by MagoKimbra

jgauchia, 2016-06-04:
Thx!!! 
Sorry for the other Issue