MKFirmware/MarlinKimbra #144: ENABLE SERVOS ISSUE

jgauchia, 2016-06-04:
When #ENABLE_SERVOS whas defined (from 4.2.82 to 4.2.84) sanitycheck raise this error:

_sketch/module/MK_Main.cpp: In function 'void homeaxis(AxisEnum)':
sketch/module/MK_Main.cpp:1659:49: error: 'deploy_z_probe' was not declared in this scope
           if (axis_home_dir < 0) deploy_z_probe();
                                                 ^
sketch/module/MK_Main.cpp:1754:26: error: 'stow_z_probe' was not declared in this scope
             stow_z_probe();
                          ^
sketch/module/MK_Main.cpp: In function 'void gcode_M401()':
sketch/module/MK_Main.cpp:7094:20: error: 'deploy_z_probe' was not declared in this scope
     deploy_z_probe();
                    ^
sketch/module/MK_Main.cpp: In function 'void gcode_M402()':
sketch/module/MK_Main.cpp:7107:25: error: 'stow_z_probe' was not declared in this scope
       stow_z_probe(false);
                         ^_

If #ENABLE_SERVO was undefined it's seems to compile correct

jgauchia, 2016-06-04:
Sorry It's because AUTO_BED_LEVELING was undefined. I'm close this isue

closed 2016-06-04 by jgauchia

MagoKimbra, 2016-06-04:
Ok i put check AUTO_BED_LEVELING if enabled servos endstop in sanity check...
Thx

jgauchia, 2016-06-04:
thx again!