MKFirmware/MarlinKimbra #145: AUTO BED LEVELING ISSUE (2) - gcode_G28()

jgauchia, 2016-06-04:
From 4.2.82 to 4.2.84
When #define MIN_Z_HEIGHT_FOR_HOMING  and undefined Z_RAISE_BEFORE_HOMING compile error.

_sketch/module/MK_Main.cpp: In function 'void gcode_G28()':
sketch/module/MK_Main.cpp:3884:37: error: 'Z_RAISE_BEFORE_HOMING' was not declared in this scope
             destination[Z_AXIS] = -(Z_RAISE_BEFORE_HOMING) * home_dir(Z_AXIS);  // Set destination away from bed_

jgauchia, 2016-06-04:
Maybe changing Z_RAISE_BEFORE_HOMING with MIN_Z_HEIGHT_FOR_HOMING in MK_Main works. I'll try it

MagoKimbra, 2016-06-04:
Ok i fix it!!!
Thx....

closed 2016-06-04 by MagoKimbra

jgauchia, 2016-06-04:
Well I've changed but still compile errors, Can you reopen the issue and post them?
I'm current testing this part of firmware , If you need some feedback from my..

MagoKimbra, 2016-06-04:
Chek now!

jgauchia, 2016-06-04:
Checked. New Issue for me , sorry :(