MKFirmware/MarlinKimbra #146: Z_PROBE

jgauchia, 2016-06-04:
I redefine Autolevel and Z_PROBE pin configuration to adapt it to the new version of firmware.
I defined Pin for Probe, etc....
But sanitycheck raise this error: 

_sketch/module/sanitycheck.h:1628:12: error: #error DEPENDENCY ERROR: You must have SERVO_ENDSTOP_ANGLES EXIST for Z Extend and Retract to use Z_PROBE_ENDSTOP.
           #error DEPENDENCY ERROR: You must have SERVO_ENDSTOP_ANGLES EXIST for Z Extend and Retract to use Z_PROBE_ENDSTOP._

Where I define the SERVO_ENDSTOP_ANGLES? I search it in code but nothing appears, only the sanitycheck code error.

Thxs!

jgauchia, 2016-06-04:
This is my configuration of R/C Servo

**/***********************************************************************
 **************************** R/C Servo ********************************
 ***********************************************************************/
#define ENABLE_SERVOS
#define Z_PROBE_ENDSTOP
#define NUM_SERVOS 1
#define Z_ENDSTOP_SERVO_NR 0
#define DEACTIVATE_SERVOS_AFTER_MOVE
#define Z_ENDSTOP_SERVO_ANGLES {70,155} // Z Axis Extend and Retract angles
/***********************************************************************/
**

jgauchia, 2016-06-04:
Maybe I need to put #define SERVO_ENDSTOP_ANGLES correct?

MagoKimbra, 2016-06-04:
Fix... Sorry...

closed 2016-06-04 by MagoKimbra

jgauchia, 2016-06-04:
Thanks!
Now it's time to DUAL_X_CARRIAGE 
I'm still working :)