MKFirmware/MarlinKimbra #151: MKDUE NOT COMPILE, i don't touch nothing!

fbarcenas, 2016-06-07:
I only put:
#define MOTHERBOARD BOARD_ULTRATRONICS

and select:
Tool-> Board-> Arduino Due (Programming port)

and....

In file included from sketch/module/HAL/HAL.h:56:0,
                 from sketch/base.h:52,
                 from /Users/Fran/Downloads/MarlinKimbra-master/MK/MK.ino:213:
sketch/module/HAL/fastio.h:31:20: fatal error: avr/io.h: No such file or directory
 #include <avr/io.h>
                    ^
compilation terminated.
exit status 1
Error compilación en tarjeta Arduino Due (Programming Port).

I am use Arduino 1.6.9, another version is better?

MagoKimbra, 2016-06-08:
You must use MK4due for Ultratronics!!!!

fbarcenas, 2016-06-08:
OMG you are truth!!!!
big fail
now i have a new fail:

 from /Users/Fran/Downloads/MarlinKimbra4due-master/MK4due/MK4due.ino:212:
sketch/module/sanitycheck.h:1594:6: error: #error DEPENDENCY ERROR: At least one of the ?_ENDSTOP_SERVO_NR is required for DEACTIVATE_SERVOS_AFTER_MOVE.
     #error DEPENDENCY ERROR: At least one of the ?_ENDSTOP_SERVO_NR is required for DEACTIVATE_SERVOS_AFTER_MOVE.
      ^

Only happen when activate:
#define DEACTIVATE_SERVOS_AFTER_MOVE

Config of servo:
#define ENABLE_SERVOS
#define NUM_SERVOS 1
#define X_ENDSTOP_SERVO_NR -1
#define Y_ENDSTOP_SERVO_NR -1
#define Z_ENDSTOP_SERVO_NR 0
#define X_ENDSTOP_SERVO_ANGLES {0,0}  // X Axis Extend and Retract angles
#define Y_ENDSTOP_SERVO_ANGLES {0,0}  // Y Axis Extend and Retract angles
#define Z_ENDSTOP_SERVO_ANGLES {90,0} // Z Axis Extend and Retract angles
#define DEACTIVATE_SERVOS_AFTER_MOVE
#define SERVO_DEACTIVATION_DELAY 300

Some idea?

closed 2016-06-10 by MagoKimbra