MKFirmware/MarlinKimbra #163: Home Issue

Dougwar, 2016-08-13:
After touch the endstop the printer move down 47mm
https://www.youtube.com/watch?v=W9Ye9hsK7ro

Do you have any suggestion?

With old fw version it works normal

Thanks

MagoKimbra, 2016-08-13:
This is normal... I've added this in new release to protect against movements x y when it is on the home.

Dougwar, 2016-08-13:
If I print a part with full Z, when its finish, it will get collided with the part?

MagoKimbra, 2016-08-13:
In the Delta you can not print in full z, unless you print a point ... When in the home to make X and Y movements of the towers should be able to get on and off, so home you can not move the X and Y. the height where it stops after the home is the real maximum height allowed to print ...

Dougwar, 2016-08-13:
It's possible to turn off it, comment a line?

Thanks

MagoKimbra, 2016-08-13:
Why?? It 'a security to avoid breaking the printer ...
The more of that value can not be printed ..