MKFirmware/MarlinKimbra #167: doesn't compile anymore when servo option activated

Windaub57, 2016-12-28:
sketch\src\MK_Main.cpp: In function 'void servo_init()':

sketch\src\MK_Main.cpp:813:5: error: 'servo' was not declared in this scope

     servo[0].attach(SERVO0_PIN);

     ^

sketch\src\MK_Main.cpp:848:18: error: 'STOW_Z_SERVO' was not declared in this scope

     STOW_Z_SERVO();

                  ^

sketch\src\MK_Main.cpp: In function 'bool set_probe_deployed(bool)':

sketch\src\MK_Main.cpp:1837:7: error: 'servo' was not declared in this scope

       servo[Z_ENDSTOP_SERVO_NR].move(z_servo_angle[deploy ? 0 : 1]);

       ^

exit status 1

Windaub57, 2016-12-28:
arduino mega 2560

#define NUM_SERVOS 1

Board?
Num servos you define?

-

MagoKimbra, 2016-12-28:
And Z_ENDSTOP_SERVO_NR ?

Windaub57, 2016-12-29:
#define Z_ENDSTOP_SERVO_NR 0

Windaub57, 2016-12-29:
looks like I opened my ticket under the wrong firmware version. I'm using the latest one : Version 4.3.09_dev

Windaub57, 2016-12-29:
sorry

MagoKimbra, 2016-12-29:
Aaaa ok....

MagoKimbra, 2016-12-29:
Do you have ramps 1.4?
I checked width ramps and all ok!!

Windaub57, 2016-12-29:
yes I have ramps 1.4. I also tried the online configurator and got the exact same pb. same after upgrading to arduino1.8.

I'll try again.

Windaub57, 2016-12-29:
New try

I took a brand fresh copy.

I only changed :
cartesian -> delta
#define Z_ENDSTOP_SERVO_NR 0
#define NUM_SERVOS 1
#define Z_PROBE_PIN         ORIG_Z_MIN_PIN

=> compil

Still get the same error. Any last ideas :) ?

MagoKimbra, 2016-12-29:
and #define ENABLE_SERVOS?