MKFirmware/MarlinKimbra #168: bug on high speed z axes

setar, 2017-01-24:
i found strange bug, see the video:
http://www.youtube.com/watch?v=ZRhAlPOKss8
first 40 sec is normal work on z=12 mm/s , and last i try give more speed (20 mm/s).
RAMPS 1.4 on Mega2560, cartesian , current dev (450bcb207a58b2cdcbdf20a443bc57fc55408a2e) with my config. 
Z_LATE is disabled

Any idea ? 
is 100% not mechanical problem.

MagoKimbra, 2017-01-25:
O_O very strange...

MagoKimbra, 2017-01-25:
Uhm... You have acc for Z at 20... Please set it to 100 and try the same test..

setar, 2017-01-27:
too same
http://www.youtube.com/watch?v=uPPDmF-NbLg

MagoKimbra, 2017-01-27:
Can you do a test with Marlin RC8?
I have to do the bench testing of an motor...
Send me the z setting, step, feedrate, acc.

setar, 2017-01-27:
test with Marlin 1.1.0-RC8
http://www.youtube.com/watch?v=1IlBMW0srQE

i think it electrical problem, too high step rate make power shutdown ( possible ATX power protection?)
with other driver shield situation analogical

base settings:
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 53.5, 53.5, 200*32/2, 300 , 300, 300 }
#define DEFAULT_MAX_FEEDRATE          { 200, 200, 12, 300, 300, 300 }
#define DEFAULT_MAX_ACCELERATION      { 800, 800, 100, 10000, 10000, 10000 }

MagoKimbra, 2017-01-27:
With M92 you set axis step per unit and 3200 or 2000 is absolutly normally for cartesian.
This problem is different with MK, and i don not know why!!
You can try disabled SLOWDOWN in your configuration please...

setar, 2017-01-27:
change driver microstep to 1/16 completely solved the problem (in  Marlin 1.1.0-RC8 )
steprate barier ?
try 1/16 microstep on current dev ...

MagoKimbra, 2017-01-27:
Yes is possible, but the steprate barrier non reset the board...

setar, 2017-01-27:
i test current dev with both disabled Z_LATE_ENABLE and SLOWDOWN
http://www.youtube.com/watch?v=X1Q9R9uqpRk

With 1/16 microstep both firmware (Marlin & MarlinKimbra)  work fine and very fast:
http://www.youtube.com/watch?v=Hd2E3MGAmyQ

MagoKimbra, 2017-01-27:
ok very good...

setar, 2017-01-28:
is not power protection, board rebooted also if ATX not switched on
maybe this the result of using options USE_WATCHDOG

setar, 2017-02-09:
hi 
i reopen the bug, and add new information:

This error affects all the axis:
(Video recorded my friends)
https://youtu.be/jpH3hNGZwbQ

it has the following settings/hardware: 
arduino due + RADDS , Hbot cinematics, X/Y=320 step/mm 
Use ENABLE_HIGH_SPEED_STEPPING does not correct the problem.

test code:
G28 X0 Y0
G1 X95 Y5 f5000
G2 F10800 X95 Y5 I5 J95
G2 F12000 X95 Y5 I5 J95
G2 F13200 X95 Y5 I5 J95
G2 F14400 X95 Y5 I5 J95
G2 F15600 X95 Y5 I5 J95
G2 F16800 X95 Y5 I5 J95
G2 F18000 X95 Y5 I5 J95
G2 F19200 X95 Y5 I5 J95
G2 F20000 X95 Y5 I5 J95
G28 Y0 X0

P.S. Earlier versions did not have this problem (when it is emerged - unknown)

P.S.S. Repieter firmware on this test work fine 
https://youtu.be/8T9sq_zOGzo

setar, 2017-02-09:
oops
I wrote the wrong place - it's all about the current version of MK4Duo

sh4jonny, 2017-04-02:
omg huge work for this test gcode. i have it too. any idea ?

MagoKimbra, 2017-04-02:
#sh4jonny use MK4duo non this version..

sh4jonny, 2017-04-03:
#MagoKimbra sorry of course I use MK4duo from master branch