MKFirmware/MarlinKimbra #170: Errori compilazione Marlin Kimbra due

mrpasquale970, 2017-04-27:
Salve. Sotto consiglio di una persona ho deciso di provare questo firmware. Però, compilandolo con Arduino ottengo i seguenti errori:

In file included from Configuration_Store.cpp:39:
/base.h:14:24: error: src/macros.h: No such file or directory
/base.h:15:23: error: src/types.h: No such file or directory
/base.h:17:27: error: src/mechanics.h: No such file or directory
/base.h:23:9: error: missing binary operator before token "("
/base.h:25:11: error: missing binary operator before token "("
/base.h:27:11: error: missing binary operator before token "("
/base.h:29:11: error: missing binary operator before token "("
/base.h:31:11: error: missing binary operator before token "("
/base.h:33:11: error: missing binary operator before token "("
/base.h:35:11: error: missing binary operator before token "("
/base.h:43:12: error: missing binary operator before token "("
/base.h:51:30: error: src/conditionals.h: No such file or directory
/base.h:52:29: error: src/sanitycheck.h: No such file or directory
/base.h:53:25: error: src/HAL/HAL.h: No such file or directory
/base.h:54:22: error: src/enum.h: No such file or directory
/base.h:56:12: error: missing binary operator before token "("
In file included from /base.h:60,
                 from Configuration_Store.cpp:39:
/Configuration_Store.h:29:15: error: missing binary operator before token "("
/Configuration_Store.h:37:14: error: missing binary operator before token "("
/Configuration_Store.h:45:14: error: missing binary operator before token "("
In file included from Configuration_Store.cpp:39:
/base.h:62:35: error: src/language/language.h: No such file or directory
/base.h:63:43: error: src/printcounter/printcounter.h: No such file or directory
/base.h:64:25: error: src/MK_Main.h: No such file or directory
/base.h:65:33: error: src/planner/planner.h: No such file or directory
/base.h:66:34: error: src/endstop/endstops.h: No such file or directory
/base.h:67:32: error: src/motion/stepper.h: No such file or directory
/base.h:68:45: error: src/motion/cartesian_correction.h: No such file or directory
/base.h:69:41: error: src/temperature/temperature.h: No such file or directory
/base.h:70:34: error: src/sensor/flowmeter.h: No such file or directory
/base.h:71:30: error: src/lcd/ultralcd.h: No such file or directory
/base.h:72:28: error: src/lcd/buzzer.h: No such file or directory
/base.h:73:37: error: src/nextion/Nextion_lcd.h: No such file or directory
/base.h:74:31: error: src/sd/cardreader.h: No such file or directory
/base.h:75:29: error: src/servo/servo.h: No such file or directory
/base.h:76:35: error: src/watchdog/watchdog.h: No such file or directory
/base.h:77:31: error: src/blinkm/blinkm.h: No such file or directory
/base.h:83:8: error: missing binary operator before token "("
/base.h:87:12: error: missing binary operator before token "("
/base.h:91:12: error: missing binary operator before token "("
Configuration_Store.cpp:178:11: error: missing binary operator before token "("
Configuration_Store.cpp:186:14: error: missing binary operator before token "("
Configuration_Store.cpp:196:12: error: missing binary operator before token "("
Configuration_Store.cpp:566:14: error: missing binary operator before token "("
Configuration_Store.cpp:590:16: error: missing binary operator before token "("
Configuration_Store.cpp:601:11: error: missing binary operator before token "("
Configuration_Store.cpp:617:14: error: missing binary operator before token "("
Configuration_Store.cpp:621:10: error: missing binary operator before token "("
Configuration_Store.cpp:625:11: error: missing binary operator before token "("
Configuration_Store.cpp:646:14: error: missing binary operator before token "("
Configuration_Store.cpp:658:10: error: missing binary operator before token "("
Configuration_Store.cpp:662:14: error: missing binary operator before token "("
Configuration_Store.cpp:674:14: error: missing binary operator before token "("
Configuration_Store.cpp:680:14: error: missing binary operator before token "("
Configuration_Store.cpp:686:14: error: missing binary operator before token "("
Configuration_Store.cpp:692:14: error: missing binary operator before token "("
Configuration_Store.cpp:709:14: error: missing binary operator before token "("
Configuration_Store.cpp:718:13: error: missing binary operator before token "("
Configuration_Store.cpp:997:10: error: missing binary operator before token "("
Configuration_Store.cpp:1004:12: error: missing binary operator before token "("
In file included from /base.h:60,
                 from Configuration_Store.cpp:39:
Configuration_Store.h:32: error: expected constructor, destructor, or type conversion before 'void'
Configuration_Store.h:33: error: expected constructor, destructor, or type conversion before 'void'
Configuration_Store.h:40: error: expected constructor, destructor, or type conversion before 'void'
Configuration_Store.h:41: error: expected constructor, destructor, or type conversion before 'void'
Configuration_Store.h:53: error: expected constructor, destructor, or type conversion before 'void'
Configuration_Store.cpp: In function 'void _EEPROM_writeData(int&, uint8_t*, uint8_t)':
Configuration_Store.cpp:147: error: 'eeprom_write_byte' was not declared in this scope
Configuration_Store.cpp:148: error: 'eeprom_read_byte' was not declared in this scope
Configuration_Store.cpp:150: error: 'ER' was not declared in this scope
Configuration_Store.cpp:150: error: 'SERIAL_LM' was not declared in this scope
Configuration_Store.cpp: In function 'void _EEPROM_readData(int&, uint8_t*, uint8_t)':
Configuration_Store.cpp:160: error: 'eeprom_read_byte' was not declared in this scope
Configuration_Store.cpp: In function 'void Config_Postprocess()':
Configuration_Store.cpp:173: error: 'planner' was not declared in this scope
Configuration_Store.cpp:189: error: 'calculate_volumetric_multipliers' was not declared in this scope
Configuration_Store.cpp:192: error: 'i' was not declared in this scope
Configuration_Store.cpp:192: error: 'LOOP_XYZ' was not declared in this scope
Configuration_Store.cpp:192: error: expected `;' before 'update_software_endstops'
Configuration_Store.cpp: In function 'void Config_ResetDefault()':
Configuration_Store.cpp:578: error: 'planner' was not declared in this scope
Configuration_Store.cpp:584: error: 'planner' was not declared in this scope
Configuration_Store.cpp:588: error: 'HOTENDS' was not declared in this scope
Configuration_Store.cpp:594: error: 'hotend_offset' was not declared in this scope
Configuration_Store.cpp:594: error: 'X_AXIS' was not declared in this scope
Configuration_Store.cpp:595: error: 'Y_AXIS' was not declared in this scope
Configuration_Store.cpp:596: error: 'Z_AXIS' was not declared in this scope
Configuration_Store.cpp:607: error: 'planner' was not declared in this scope
Configuration_Store.cpp:614: error: 'home_offset' was not declared in this scope
Configuration_Store.cpp:614: error: 'X_AXIS' was not declared in this scope
Configuration_Store.cpp:614: error: 'Y_AXIS' was not declared in this scope
Configuration_Store.cpp:614: error: 'Z_AXIS' was not declared in this scope
Configuration_Store.cpp:706: error: 'volumetric_enabled' was not declared in this scope
Configuration_Store.cpp:714: error: 'SERIAL_EM' was not declared in this scope
Configuration_Store.cpp: In function 'void ConfigSD_ResetDefault()':
Configuration_Store.cpp:999: error: 'print_job_counter' was not declared in this scope
Configuration_Store.cpp:1000: error: 'OK' was not declared in this scope
Configuration_Store.cpp:1000: error: 'SERIAL_LM' was not declared in this scope

MagoKimbra, 2017-04-27:
Bisogna usare l'ultima versione di ide.. Le versioni vecchio non contemplano l'uso di sottocartelle per i file.. Quindi non li trova...
/base.h:14:24: error: src/macros.h: No such file or directory
/base.h:15:23: error: src/types.h: No such file or directory

mrpasquale970, 2017-04-27:
Perfetto ti ringrazio

mrpasquale970, 2017-04-27:
Mh, ho parlato presto. il firmware viene caricato ma a quanto pareho sbagliato ad inserire qualche parametro online, l'lcd mi fischia e il motore dell'asse x cammina a scatti da solo. Ho una Geeetech i3 Pro B..per caso c'è qualche configurazione già testata da cui poter partire? Ho mancato qualche informazione per avere una buona risposta?

MagoKimbra, 2017-04-27:
No, che scheda hai messo? Sai come è nel firmware che ti danno?

mrpasquale970, 2017-04-27:
La scheda è della casa "GT 2560", dunque è una copia di Arduino atmega 2560. Il vecchio firmware era il Marlin a cui ho modificato qualche parametro (davvero nulla di che) di cui ho caricato anche il Configuration.h online per scaricare MK ma sembra non essere servito. L' LCD è uno smart 2004, non trovandolo nell'elenco ho scelto "lcd reprapdiscount" che mi sembrava lo stesso

MagoKimbra, 2017-04-27:
Ok ma perchè hai scritto marlinkimbra 2??

mrpasquale970, 2017-04-27:
Chiedo scusa è semplicemente questo: http://marlinkimbra.it/configurator/v4_2_9/firmware_configurator.php
Anche per me avrebbe dovuto funzionare senza problemi, ho usato ovviamente il configuratore. Ma il fatto che il suddetto configuratore si apra nonostante abbia inserito il configuration.h vecchio è normale? Ho pensato potesse essere un problema di comunicazione ai pin ma non ho modificato nulla (tranne che l'asse x fosse invertito)  circa questa cosa quindi non dovrei avere problemi in teoria..
non posso nemmeno verificare se il problema è solamente riferito a solo questo motore perché l'LCD manco a dirlo non  funziona oltre che a fare un bip fastidiosissimo