MKFirmware/MarlinKimbra #174: avr/io.h missing

jmlcatarino, 2017-07-17:
Hi there, I'm trying to upload this firmware to a Due and it gives me this error : sketch/src/HAL/fastio.h:31:20: fatal error: avr/io.h: No such file or directory
 #include <avr/io.h>

I can't find any solutions for this error. I was using the earlier version, MK4Due and it gave lot's of errors with LCD settings, by not recognizing several parts of the code, so I'm trying this new version. Using IDE 1.8.3
Thank you in advance

MagoKimbra, 2017-07-17:
This Firmware is old and only for 8 bit processor...
Look https://github.com/MKFirmware/MK4duo

jmlcatarino, 2017-07-17:
I thought it was the newer version. I went to your site and got the idea that it was newer than MK4duo. :) Thanks anyway and a special thanks for the firmware.

closed 2017-08-24 by jmlcatarino