MKFirmware/MarlinKimbra #177: SD Card Question

CJM101, 2017-10-26:
I'm new to Marlin as well as the Nextion display. So please work with me

Is it possible to use the nextion's SD Card slot for loading files?

I ask because its not working for me. I have SDSUPPORT enabled and I can go into the SD Card directory, but theres nothing there, its not listing my .gcode files.

My setup:
Arduino Mega 2560
Ramps V1.4
Nextion 4.3" display

MagoKimbra, 2017-10-26:
No, you can not use it. The next board sd card only serves to upgrade the firmware of the display, it is not seen in any way from the arduino mega board. You have to put the sd module for ramps.
http://reprap.org/mediawiki/images/thumb/b/b9/Connectsdramp1.jpg/500px-Connectsdramp1.jpg

CJM101, 2017-10-28:
Thanks for the help and I apologize for this post. I ran across another thread last night that would've answered my question. Just didn't find it before posting this question.

I also just want to say that I think what You guys have managed to pull off with marlin and nextion is very impressive. The interface is awesome and the coding is no less than amazing.

closed 2017-10-28 by CJM101