MKFirmware/MarlinKimbra #178: Laser Cutter slow down during tight turns or Circles

prusanoff, 2017-12-19:
Hi all,
During cutting my laser slows down or even stops while cutting tight circles or curves. The effect increases, as the nominal size of the circle decreases (ex. with radii of 10 it just slows down, with radii of 3mm it stops and produces strange noise). The resulting feature is hardly approximate a circle and there is dark spots along the circumference, also step-loss occurs. I've already excluded every possible mechanical (like backlash, lubrication, etc.) or electrical source (driver/soldering, etc. problems), the machine has been calibrated with professional measurement tools and every part of it is a CNC machined metal or industrial grade part.
Machine type: experimental cartesian printer
Controller: MKS Gen V1.4
MarlinKimbra Latest and up to date 
Host: LaserWeb

 motion settings:
Speed:   900 mm/s, Acceleration: 12 000 mm/s^2, "Jerk": 50 mm/s
 
Some ideas/thoughts:
 So I have googled the hell out of this and someone using marlin had the same problem [Here](https://github.com/MarlinFirmware/Marlin/issues/6023#issue-214003986). I have tried the latest version of marlin on my board and it runs nice and smooth through the turns, curves, or circles. The issue is only MarlinKimbra. I love laserWeb and its direct connection however I can't seem to figure this out 

Here is a list of thing I have tried 

Older version or MarlinKimbra for my laser:

`You can edit "board.txt" in arduino ide, copy the "mega 2560" board declaration, call it 
"mega 2560 serial buffer 256" and then add a line like:

mega256.build.extra_flags=-DSERIAL_RX_BUFFER_SIZE=256 -DSERIAL_TX_BUFFER_SIZE=256`

Small effect it took quarter turns on the small holes but still stuttered like hell 

Zero Effect from here on out 

- turn-off autobed-leveling : no effect
- changing jerk: no effect
- using SD is not possible, but I've also tried different baudrates: no effect.
- I've tried different settings (very low speed/acc/jerk): no effect.

Here are the setting I have played with and have had no affect 
```

#define FASTER_GCODE_PARSER
#define SLOWDOWN
#define BABYSTEP_XY 

#define DEFAULT_AXIS_STEPS_PER_UNIT   {157.4802, 157.4802, 3200, 625, 625, 625, 625}
/*****************************************************************************************
 ********************************** Axis feedrate ****************************************
 *****************************************************************************************/
#define DEFAULT_MAX_FEEDRATE          {7600, 7600, 10, 25, 0, 0, 0}
#define MANUAL_FEEDRATE               {7600, 7600, 10, 25}
#define DEFAULT_MINIMUMFEEDRATE       0.0
#define DEFAULT_MINTRAVELFEEDRATE     0.0
#define MINIMUM_PLANNER_SPEED         0.05                       // (mm/sec)
 //*****************************************************************************************/
#define DEFAULT_MAX_ACCELERATION              {3000, 3000, 50, 1000, 1000, 1000, 1000}
#define DEFAULT_RETRACT_ACCELERATION          {2000, 2000, 2000, 2000}
#define DEFAULT_ACCELERATION          3000

#define DEFAULT_TRAVEL_ACCELERATION   3000
//*****************************************************************************************/
#define DEFAULT_XJERK 10.0
#define DEFAULT_YJERK 10.0
#define DEFAULT_ZJERK  0.4
#define DEFAULT_EJERK                   {5.0, 5.0, 5.0, 5.0}
 

#define MICROSTEP_MODES {16, 16, 16, 16}

```

MagoKimbra, 2017-12-19:
What commands do you use to make curves? G2 G3?

prusanoff, 2018-01-09:
I am taking a guess here I think the issue is that I am using a MKS Gen board board witch is not supported by MK