MKFirmware/MarlinKimbra #179: TMC2208 does not work on DUE!

pirat777knj, 2017-12-24:
Hey.
I decided to put 2208 on RADDS, but errors are made when compiling.
The compiler warns that the library does not work on x86 (((
That is, on 32 bits
I would like to see support for these drivers)))

> `ПРЕДУПРЕЖДЕНИЕ: библиотека TMC2208Stepper должна запускаться на архитектурах (avr) и может быть несовместима с вашей платой на архитектуре (sam).
> sketch\src\core\stepper\stepper_indirection.cpp:247:40: error: 'SoftwareSerial' does not name a type
> 
> #define _TMC2208_DEFINE_SOFTWARE(ST) SoftwareSerial stepper##ST##_serial = SoftwareSerial(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN); \
> 
>                                     ^
> sketch\src\core\stepper\stepper_indirection.cpp:255:7: note: in expansion of macro '_TMC2208_DEFINE_SOFTWARE'
> 
>    _TMC2208_DEFINE_SOFTWARE(X);
> 
>    ^
> sketch\src\core\stepper\stepper_indirection.cpp:248:68: error: 'stepperX_serial' was not declared in this scope
> 
>                                     TMC2208Stepper stepper##ST(&stepper##ST##_serial, ST##_SERIAL_RX_PIN > -1)
> 
>                                                                 ^
> sketch\src\core\stepper\stepper_indirection.cpp:255:7: note: in expansion of macro '_TMC2208_DEFINE_SOFTWARE'
> 
>    _TMC2208_DEFINE_SOFTWARE(X);
> 
>    ^
> sketch\src\core\stepper\stepper_indirection.cpp:247:40: error: 'SoftwareSerial' does not name a type
> 
> #define _TMC2208_DEFINE_SOFTWARE(ST) SoftwareSerial stepper##ST##_serial = SoftwareSerial(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN); \
> 
>                                     ^
> sketch\src\core\stepper\stepper_indirection.cpp:269:7: note: in expansion of macro '_TMC2208_DEFINE_SOFTWARE'
> 
>    _TMC2208_DEFINE_SOFTWARE(Y);
> 
>    ^
> sketch\src\core\stepper\stepper_indirection.cpp:248:68: error: 'stepperY_serial' was not declared in this scope
> 
>                                     TMC2208Stepper stepper##ST(&stepper##ST##_serial, ST##_SERIAL_RX_PIN > -1)
> 
>                                                                 ^
> sketch\src\core\stepper\stepper_indirection.cpp:269:7: note: in expansion of macro '_TMC2208_DEFINE_SOFTWARE'
> 
>    _TMC2208_DEFINE_SOFTWARE(Y);
> 
>    ^
> sketch\src\core\stepper\stepper_indirection.cpp:247:40: error: 'SoftwareSerial' does not name a type
> 
> #define _TMC2208_DEFINE_SOFTWARE(ST) SoftwareSerial stepper##ST##_serial = SoftwareSerial(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN); \
> 
>                                     ^
> sketch\src\core\stepper\stepper_indirection.cpp:276:7: note: in expansion of macro '_TMC2208_DEFINE_SOFTWARE'
> 
>    _TMC2208_DEFINE_SOFTWARE(Y2);
> 
>    ^
> sketch\src\core\stepper\stepper_indirection.cpp:248:68: error: 'stepperY2_serial' was not declared in this scope
> 
>                                     TMC2208Stepper stepper##ST(&stepper##ST##_serial, ST##_SERIAL_RX_PIN > -1)
> 
>                                                                 ^
> sketch\src\core\stepper\stepper_indirection.cpp:276:7: note: in expansion of macro '_TMC2208_DEFINE_SOFTWARE'
> 
>    _TMC2208_DEFINE_SOFTWARE(Y2);
> 
>    ^
> sketch\src\core\stepper\stepper_indirection.cpp:247:40: error: 'SoftwareSerial' does not name a type
> 
> #define _TMC2208_DEFINE_SOFTWARE(ST) SoftwareSerial stepper##ST##_serial = SoftwareSerial(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN); \
> 
>                                     ^
> sketch\src\core\stepper\stepper_indirection.cpp:283:7: note: in expansion of macro '_TMC2208_DEFINE_SOFTWARE'
> 
>    _TMC2208_DEFINE_SOFTWARE(Z);
> 
>    ^
> sketch\src\core\stepper\stepper_indirection.cpp:248:68: error: 'stepperZ_serial' was not declared in this scope
> 
>                                     TMC2208Stepper stepper##ST(&stepper##ST##_serial, ST##_SERIAL_RX_PIN > -1)
> 
>                                                                 ^
> sketch\src\core\stepper\stepper_indirection.cpp:283:7: note: in expansion of macro '_TMC2208_DEFINE_SOFTWARE'
> 
>    _TMC2208_DEFINE_SOFTWARE(Z);
> 
>    ^
> sketch\src\core\stepper\stepper_indirection.cpp:247:40: error: 'SoftwareSerial' does not name a type
> 
> #define _TMC2208_DEFINE_SOFTWARE(ST) SoftwareSerial stepper##ST##_serial = SoftwareSerial(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN); \
> 
>                                     ^
> sketch\src\core\stepper\stepper_indirection.cpp:297:7: note: in expansion of macro '_TMC2208_DEFINE_SOFTWARE'
> 
>    _TMC2208_DEFINE_SOFTWARE(E0);
> 
>    ^
> sketch\src\core\stepper\stepper_indirection.cpp:248:68: error: 'stepperE0_serial' was not declared in this scope
> 
>                                     TMC2208Stepper stepper##ST(&stepper##ST##_serial, ST##_SERIAL_RX_PIN > -1)
> 
>                                                                 ^
> sketch\src\core\stepper\stepper_indirection.cpp:297:7: note: in expansion of macro '_TMC2208_DEFINE_SOFTWARE'
> 
>    _TMC2208_DEFINE_SOFTWARE(E0);
> 
>    ^`
>

MagoKimbra, 2017-12-24:
sorry for 2208 0n 32BIT only hardware serial supportated!!!

pirat777knj, 2017-12-24:
Ok. but how can they be connected?

MagoKimbra, 2017-12-24:
On serial hardware Serial 1, Serial 2 or Serial 3.
http://www.robgray.com/temp/Due-pinout-A4.png
Pin 14 15, 16 17 or 18 19.

pirat777knj, 2017-12-25:
Thank you, though it does not completely solve the question.
Drivers 5 ...