MKFirmware/MarlinKimbra #180: z probe servo not work anet a8

ISU000, 2017-12-25:
hi, i test in all input 27,28,29,30 but with m280 don't move servo (3D Touch) the 3D Touch with marlin 3.1.6 in my anet work well but without nextion 7 lcd.

This is my configuration:
[Configuration_Overall.h.zip](https://github.com/MKFirmware/MarlinKimbra/files/1585932/Configuration_Overall.h.zip)

MagoKimbra, 2017-12-27:
#define SERVO0_PIN 30 in configuration_overall????? WHY??? The pin is defined in configuration_pins!!

ISU000, 2017-12-27:
Yes i know but in original input 27 in board 100 anet when i put servo cable de light 1 go on, for test i change inputs in board configuration 27,28,29,30 and 30 servo dont move, i test with normal servo and not move. In marlin 3.1.6 with anet board work perfect.

I put in overall for make test but servo not work.
This moning i test with ramps 1.4 board and work perfect. 

Thanks
Iker

MagoKimbra, 2017-12-27:
Ok you must change on your board file

ISU000, 2017-12-27:
Yes