MKFirmware/MarlinKimbra #181: Issue with thermal protection

EPES-Rades, 2018-02-02:
I would like to ask you about issue thermal protection. For example: If I disconnect thermisto of bed, printer write:

Error:MINTEMP triggered, system stopped! Heater_ID: Bed
[ERROR] Error:MINTEMP triggered, system stopped! Heater_ID: Bed

Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!

but the bed still heating. Same issue with heater.
I tried to change below parametrs but without result:

#define INVERTED_HEATER_PINS false
#define INVERTED_BED_PIN false

#define THERMAL_PROTECTION_HOTENDS true
#define THERMAL_PROTECTION_BED true

Thank you for your answere / help.

Radek B.

MagoKimbra, 2018-02-02:
This firmware is old. Please go to https://github.com/MKFirmware/MK4duo for new firmware..

EPES-Rades, 2018-02-03:
Ok, thank you for answer.
I have downloaded the newest firmware V4.3.4. I'll try it. I had the firmware V4.3.2.

I'm sorry. I choose wrong "Issues" branch.

closed 2018-02-04 by MagoKimbra