MKFirmware/MarlinKimbra #186: problem on execute command M303 E-1 S60 C8

demonbd, 2018-05-09:
Hi , on execute command M303 E-1 S60 C8 , dont' heat yhe bed but the tool E0.

use Last firmware 
[MK4duo.zip](https://github.com/MKFirmware/MarlinKimbra/files/1987093/MK4duo.zip)

MagoKimbra, 2018-05-09:
E0 is extruder!! You want to do the pid on an extruder ?? You'll want to do it on a hotend or on the bed and those are heaters so the command is H-1 not E-1

demonbd, 2018-05-09:
the command it's foun on repram wiki
In the case of multiple extruders (E0, E1, E2) these PID values are shared between the extruders, although the extruders may be controlled separately. If the EEPROM is enabled, save with M500. If it is not enabled, save these settings in Configuration.h.

For the bed, use:

 Bed: M303 E-1 S60 C8 

http://reprap.org/wiki/PID_Tuning
thank's 
regards

MagoKimbra, 2018-05-09:
Ok, but with this firmware the command is M303 H and not E
PID relay autotune: H[heaters] H = 0-3 Hotend, H = -1 BED, H = -2 CHAMBER, H = -3 COOLER, S sets the target temperature (default target temperature = 200C), C, R, U, R 0 = Classic Pid, 1 = Some overshoot, 2 = No Overshoot, 3 = Pessen Pid.

https://github.com/MKFirmware/MK4duo/blob/master/Documentation/GCodes.md

demonbd, 2018-05-09:
Thank's a lot

closed 2018-05-09 by demonbd