MKFirmware/MarlinKimbra #188: Endstop is always trigged

set4up, 2018-10-25:
Hey. I tried to configure the limit switches, but they are always read by the command M119 TRIGGED. I changed the settings in different ways to find when they earn correctly. Inversion does not help. By pressing the hand on the switch reads OPENED. In the firmware marlin and repeater everything works. Please help.

MagoKimbra, 2018-10-25:
This parameters is saved in eeprom in MK4duo you must change with command gcode and saved in eeprom or reset eeprom if you change configuration...

set4up, 2018-10-25:
Thank. I'll try

set4up, 2018-10-25:
Unfortunately it did not help. I reset eeprom with the command m502

MagoKimbra, 2018-10-25:
Ok.. change the pullup valor or logic with gcode and try...

set4up, 2018-10-25:
I do not know how to do it

set4up, 2018-10-25:
#define ENDSTOPPULLUPS

#if DISABLED(ENDSTOPPULLUPS)
// fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
//#define ENDSTOPPULLUP_XMIN
//#define ENDSTOPPULLUP_YMIN
//#define ENDSTOPPULLUP_ZMIN
//#define ENDSTOPPULLUP_Z2MIN
//#define ENDSTOPPULLUP_Z3MIN
//#define ENDSTOPPULLUP_Z4MIN
//#define ENDSTOPPULLUP_XMAX
//#define ENDSTOPPULLUP_YMAX
//#define ENDSTOPPULLUP_ZMAX
//#define ENDSTOPPULLUP_Z2MAX
//#define ENDSTOPPULLUP_Z3MAX
//#define ENDSTOPPULLUP_Z4MAX
//#define ENDSTOPPULLUP_ZPROBE
//#define ENDSTOPPULLUP_EMIN
#endif
/*****************************************************************************************/

/*****************************************************************************************
 ************************************ Endstops logic *************************************
 *****************************************************************************************
 *                                                                                       *
 * Mechanical endstop with COM to ground and NC to Signal                                *
 * uses "false" here (most common setup).                                                *
 *                                                                                       *
 *****************************************************************************************/
#define X_MIN_ENDSTOP_LOGIC   false   // set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_LOGIC   true   // set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_LOGIC   false   // set to true to invert the logic of the endstop.
#define Z2_MIN_ENDSTOP_LOGIC  true   // set to true to invert the logic of the endstop.
#define Z3_MIN_ENDSTOP_LOGIC  true   // set to true to invert the logic of the endstop.
#define Z4_MIN_ENDSTOP_LOGIC  true   // set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_LOGIC   true   // set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_LOGIC   true   // set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_LOGIC   true   // set to true to invert the logic of the endstop.
#define Z2_MAX_ENDSTOP_LOGIC  true   // set to true to invert the logic of the endstop.
#define Z3_MAX_ENDSTOP_LOGIC  true   // set to true to invert the logic of the endstop.
#define Z4_MAX_ENDSTOP_LOGIC  true   // set to true to invert the logic of the endstop.
#define Z_PROBE_ENDSTOP_LOGIC true   // set to true to invert the logic of the probe.
#define E_MIN_ENDSTOP_LOGIC   true   // set to true to invert the logic of the endstop.

I specifically set up different endstop in different ways to see which setting helped

MagoKimbra, 2018-10-25:
Yes ok but this does not help me, I do not know what endstop you have and how they are connected. Try using the gcode commands you do it in real time without having to redo each time ... The list of gcode commands can be found here on github or inside the .ino file.
Anyway, they are M123 and M124

set4up, 2018-10-26:
Thank. I'll try in the evening

set4up, 2018-10-26:
I found a solution:
The Endstop settings in Configuration_Cartesian.x are not affected. It needs to be configured in Configuration_O. After that it worked.
Conducted an experiment:
1. Commenting out the Endstop settings in Configuration_Overall.x. The compiler gave an error, so this setting is taken into account.
2. Commented out the Endstop settings in Configuration_Cartesian.x and everything compiled and works.
Conclusion: the Endstop settings (at least) are not taken into account when compiling the sketch.