MKFirmware/MarlinKimbra #190: Power off hotend and heatbed when print

ZhenyaKa, 2019-01-08:
Just power off both heater and bed :(((

Two different (mirrored) pcs at the same heigh.

Hardware: DUE+RAMPS-FD+LV8729 (XY) 8825 (Z+E)

Repetier, Marlin 2.0 works well

`/**
 * MK4duo Firmware for 3D Printer, Laser and CNC
 *
 * Based on Marlin, Sprinter and grbl
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 * Copyright (C) 2013 Alberto Cotronei @MagoKimbra
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program. If not, see <http://www.gnu.org/licenses/>.
 */

/**
 * Configuration_Overall.h
 * Here you can define all your custom settings and they will overwrite configurations in the main configuration files.
 */

/*******************************
 *   Firmware Version V4.3.7   *
 *******************************/

#define CONFIGURATION_OVERALL

/***********************
 * Configuration_Basic *
 ***********************/
#define SERIAL_PORT_1 0
#define BAUDRATE_1 250000
#define SERIAL_PORT_2 -1
#define BAUDRATE_2 250000
#define STRING_CONFIG_H_AUTHOR "(none, default config)"
#define MACHINE_UUID "00000000-0000-0000-0000-000000000000"
#define KILL_METHOD 0
#define NO_TIMEOUTS 1000
//#define ADVANCED_OK
//#define EMERGENCY_PARSER
#define FASTER_GCODE_PARSER
#define FASTER_GCODE_EXECUTE
#define HOST_KEEPALIVE_FEATURE
#define DEFAULT_KEEPALIVE_INTERVAL 2
#define MOTHERBOARD BOARD_RAMPS_FD_V1
#define MECHANISM MECH_COREXY
//#define MECHANISM MECH_COREXY
//#define MECHANISM MECH_COREYX
//#define MECHANISM MECH_COREXZ
//#define MECHANISM MECH_COREZX
//#define MECHANISM MECH_COREYZ
//#define MECHANISM MECH_COREZY
//#define MECHANISM MECH_DELTA
//#define MECHANISM MECH_MORGAN_SCARA
//#define MECHANISM MECH_MAKERARM_SCARA
//#define MECHANISM MECH_MUVE3D
#define POWER_SUPPLY 0
[... 1667 lines cut ...]
"angleretractservosX": 0,
"angleextendservosY": 0,
"angleretractservosY": 0,
"uilanguages": 7,
"easyload": 0,
"bowdenlenght": 250,
"lcdpurgelenght": 10,
"lcdretractlenght": 5,
"lcdpurgefeedrate": 3,
"lcdretractfeedrate": 5,
"lcdloadfeedrate": 20,
"lcdunloadfeedrate": 20,
"plahotendtemp": 190,
"plabedtemp": 60,
"plafanspeed": 255,
"abshotendtemp": 240,
"absbedtemp": 100,
"absfanspeed": 255,
"gumhotendtemp": 230,
"gumbedtemp": 60,
"gumfanspeed": 255
}
========== End configuration string ==========
*/
`

ZhenyaKa, 2019-01-08:
Forgot to say:
Print via Octoprint:

"Error reported by printer
Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: cold extrusion prevented"

MagoKimbra, 2019-01-08:
You must do the auto tune pid and save the new valor in eeprom..
Look the log when you turn on the heaters...

ZhenyaKa, 2019-01-08:
PID is ok. It turn off when printing.

I got checksum errors in LOG:

| Send: N11633 G1 X92.034 Y140.036 E2.3119*96
| Recv: ok
| Send: N11634 G1 X91.753 Y140.321 E2.3253*106
| Recv: ok
| Send: N11635 G1 X91.788 Y140.355 E2.3269*103
| Recv: Error:checksum mismatch, Last Line: 11633
| Recv: Resend:11634
2019-01-08 15:48:11,694 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 14437, current line = 14442
| Last lines in terminal:
| Recv: ok
| Send: N14433 G92 E0*70
| Send: N14434 G1 X135.649 Y98.781 E0.0211 F3900*9
| Recv: ok
| Send: N14435 G1 X135.092 Y99.225 E0.0448*100
| Recv: ok
| Send: N14436 G1 X134.503 Y99.631 E0.0685*105
| Recv: ok
| Recv: ok
| Send: N14437 G1 X133.811 Y100.038 E0.0952*90
| Recv: ok
| Send: N14438 G1 X133.180 Y100.351 E0.1187*89
| Recv: X:136.12 Y:98.35 Z:4.440 E:0.0000
| Recv: ok
| Send: N14439 G1 X132.553 Y100.604 E0.1412*95
| Send: N14440 G1 X132.051 Y100.767 E0.1587*95
| Recv: ok
| Send: N14441 G1 X131.704 Y100.938 E0.1716*84
| Recv: Error:checksum mismatch, Last Line: 14436
| Recv: Resend:14437

MagoKimbra, 2019-01-08:
When printing??
Explain turn off at start or more 30 minutes?

ZhenyaKa, 2019-01-08:
About 30 mins with errors

2019-01-08 15:32:51,943 - octoprint.util.comm - INFO - Firmware didn't send an 'ok' with their resend request. That's a known bug with some firmware variants out there. Simulating an ok to continue...
2019-01-08 15:32:53,583 - octoprint.plugins.preheat - INFO - Preheating bed to 110.0
2019-01-08 15:32:53,587 - octoprint.plugins.preheat - INFO - Preheating tool0 to 250.0
2019-01-08 15:34:07,085 - octoprint.util.comm - INFO - Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
2019-01-08 15:34:10,861 - octoprint.util.comm - INFO - Telling the printer to set the busy interval to our "communicationBusy" timeout - 1s = 2s
2019-01-08 15:34:12,089 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 51, current line = 53
| Last lines in terminal:
| Recv: ok
| Send: N48 M84*35
| Recv: X:0.00 Y:0.00 Z:0.000 E:0.0000
| Recv: ok
| Send: N49 M105*26
| Recv: busy:processing
| Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
| Recv: busy:processing
| Recv: ok
| Send: N50 M113 S2*84
| Recv: ok T:213.23 /250 B:89.79 /110 B@:255 @:255
| Recv: wait
| Send: N51 M105*19
| Recv: ok
| Recv: echo:Unknown command: ""
| Recv: ok
| Recv: wait
| Send: N52 M105*16
| Recv: Error:Line Number is not Last Line Number+1, Last Line: 50
| Recv: Resend:51
2019-01-08 15:35:35,377 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Printing"
2019-01-08 15:35:35,460 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-01-08 15:36:07,410 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2019-01-08 15:36:19,848 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2019-01-08 15:36:32,190 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2019-01-08 15:36:44,457 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2019-01-08 15:36:56,701 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2019-01-08 15:37:08,940 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2019-01-08 15:37:21,889 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2019-01-08 15:37:27,294 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2019-01-08 15:37:34,300 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2019-01-08 15:37:37,341 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 446, current line = 451
| Last lines in terminal:
| Recv: ok
| Send: N443 M105*36
| Recv: ok
| Send: N444 G1 X76.556 Y161.072 E0.2508*109
| Recv: ok
| Send: N445 G1 X76.570 Y161.058 E0.2520*106
| Recv: X:87.89 Y:147.34 Z:0.240 E:0.0000
| Recv: ok
| Send: N446 G1 X73.396 Y157.884 E0.5207*107
| Recv: ok
| Send: N447 G1 X73.712 Y157.568 E0.5475*110
| Recv: ok
| Send: N448 G1 X72.552 Y156.407 E0.6458*99
| Recv: ok
| Send: N449 G1 X72.659 Y156.299 E0.6549*106
| Recv: X:73.59 Y:158.93 Z:0.240 E:0.0000
| Recv: ok
| Send: N450 G1 X76.994 Y160.634 E1.0219*107
| Recv: Error:checksum mismatch, Last Line: 445
| Recv: Resend:446
2019-01-08 15:37:37,348 - octoprint.util.comm - INFO - Ignoring resend request for line 446, that still originates from lines we sent before we got the first resend request
2019-01-08 15:37:37,355 - octoprint.util.comm - INFO - Ignoring resend request for line 446, that still originates from lines we sent before we got the first resend request
2019-01-08 15:39:18,532 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 1588, current line = 1591
| Last lines in terminal:
| Send: N1582 G1 Z0.640 F300*21
| Recv: ok
| Send: N1583 G1 E0.0000 F2400*12
| Recv: X:127.00 Y:101.77 Z:0.440 E:0.0000
[... 120 lines cut ...]
2019-01-08 16:05:35,932 - octoprint.util.comm - WARNING - Received an error from the printer's firmware: cold extrusion prevented
| Last lines in terminal:
| Send: N34974 G1 X139.096 Y108.211 E0.4206*80
| Recv: X:144.70 Y:113.77 Z:13.440 E:0.0000
| Recv: ok
| Send: N34975 G1 X143.987 Y113.067 E0.6498*95
| Recv: ok
| Send: N34976 G1 X144.697 Y113.771 F3900*65
| Recv: ok
| Send: N34977 G1 X145.268 Y113.775 F9000*66
| Recv: ok
| Send: N34978 G92 E0*70
| Recv: ok
| Send: N34979 G1 X142.897 Y116.073 E0.1098 F3900*61
| Recv: ok
| Send: N34980 G1 X136.940 Y110.158 E0.3890*83
| Recv: ok
| Send: N34981 G1 X139.069 Y107.621 E0.4992*84
| Recv: X:145.27 Y:113.78 Z:13.440 E:0.0000
| Recv: ok
| Send: N34982 G1 X144.558 Y113.070 E0.7565*90
| Recv: Error:cold extrusion prevented
2019-01-08 16:05:35,937 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Error: cold extrusion prevented"
2019-01-08 16:05:36,089 - octoprint.util.comm - INFO - Changing monitoring state from "Error: cold extrusion prevented" to "Offline (Error: cold extrusion prevented)"
2019-01-08 16:05:37,814 - octoprint.plugin - ERROR - Error while calling plugin printhistory

MagoKimbra, 2019-01-08:
Ok the printer is not printing ...
When printing from USB, then HOST, you have to tell the printer that it is starting to print, command M530 S1.
Repetier Host or Repetier Server already do this octoprint no, so you have to put it in the start gcode.
At the end of the press there must be M530 S0 to say that you have finished printing.
If you do not, a security system turns off the heaters after 30 minutes.

ZhenyaKa, 2019-01-08:
You must include this in FAQ. It is very very hidden protection!

MagoKimbra, 2019-01-08:
Why? The normal host already gives that command is a standard, octoprint is not yet adequate, the problem is octoprint ...
However you find it written in readme.md
https://github.com/MKFirmware/MK4duo
`
Added protection of the heaters if you do not print after 30 minutes on. It is important that the host gives the command M530 S1 for the start and M530 for the stop.
`

ZhenyaKa, 2019-01-08:
Yes but usally no many people reads release notes. And this "bug" very hidden.

MagoKimbra, 2019-01-08:
Tell me where should I write it?

ZhenyaKa, 2019-01-08:
I don't know, but I think you will get a lot of "issues" from octoprint users if they want to use Kimbra and dig deeper :)

Or just switch to another FW

ZhenyaKa, 2019-01-08:
Maybe good place will be in configiration_overall.h or in web-configurator :)
ATTENTION!!!
ATTENTION!!!
ATTENTION!!!

add S503 S1 in start gcode :)

MagoKimbra, 2019-01-08:
Only if you have the normal host.. Octoprint isn't normal host...
If octoprint do not put M104 S200, for example, i must write put in start gcode M104 for setting temperature??? No i think no!!!
The command M530 for start printing is a standard in new protocol, if octoprint do not put this comand the problem is octoprint non my...

foosel, 2019-01-10:
> The command M530 for start printing is a standard in new protocol, if octoprint do not put this comand the problem is octoprint non my...

[The only thing that we have with regards to a spec](https://reprap.org/wiki/G-code#M530:_Enable_printing_mode) states this:

> This enables the firmware to switch into a special print display mode to show print progress.

It does say nothing of "firmware is also free to disable heaters without warning if this command is not sent". You can't just attach functionality as disruptive as that to new capabilities that are targeting a completely different use case and then claim host software that doesn't yet support new capabilities that are defined for purely asthetic reasons is buggy because your firmware does more than it should be doing. 

You decided to add functionality that relies on `M530` being sent even though it is merely defined for "special printing displays". Don't put the blame for issues resulting from that decision on OctoPrint.

MagoKimbra, 2019-01-10:
I did not give it fault to Octoprint, I just said that the M530 command does: This command tells the firmware that a print has started (S1) or ended (S0). Enables a range of features within firmware, counters and more including temperature protection control if you are not printing.
It is a command that hosts give. I said if octoprint does not give that command to indicate the print start, it should be put in the start gcode.
The MK4duo documentation states that if there is no M530 S1 command, the printer does not know it is printing from the host, it is only executing a series of commands and then activating the 30-minute timer for the temperature.
The user told me that I have to write it in large letters and I told him that if octoprint does not give that command I can not do anything about it and it is he who has to put it in the start gcode.

foosel, 2019-01-10:
> I did not give it fault to Octoprint

I'm sorry, but you very clearly did. You basically blamed OctoPrint for not sending this new command as default already (which, considering that it's defined *for display purposes* I don't see a pressing need considering a never ending TODO list with way more important things on it) for OP's problems and called it a "not normal host" due to that. My problem is with that. I don't like my software being blamed for problems caused by firmware (like this one) adding additional critical functionality to unrelated *optional* commands.

The problem is that you overloaded the function of `M530` and that users aren't finding that in the documentation. That's ok. But don't go declaring host software that isn't aware of some firmware derivative overloading that command with additional thermal monitoring "unnormal" because due to that decision *you* made.

MagoKimbra, 2019-01-10:
I apologize, but unfortunately speaking bad English I used inappropriate terms.

ZhenyaKa, 2019-01-10:
write this "protection" in any FAQ please both developers!

MagoKimbra, 2019-01-10:
It's is write:
Added protection of the heaters if you do not print after 30 minutes on. It is important that the host gives the command M530 S1 for the start and M530 for the stop.
https://github.com/MKFirmware/MK4duo

Where i can write this??

foosel, 2019-01-10:
> Since the hosts or servers were sending the command, I used it for this.

"Since *some* of the hosts or servers were sending the command" you mean ;)

> write this "protection" in any FAQ please both developers!

Sure. From my experience it won't be read though, I sadly have the same experiences as @MagoKimbra to that regard.

EvilGremlin, 2019-02-08:
Kimbra clearly in the wrong here. When saying "The command M530 for start printing is a standard in new protocol", you should point us to ISO 6983-1:2009. But guess what, there is no M530 in ISO standard. And any commands not mentioned in ISO works purely on a whims of independent developers. Clearly, the word "standard" isn't applicable here in any imaginable way.
And when introducing new command or rewriting old widespread ones so that by default now it will screw jobs - you should write about it explicitly in big red font everywhere, so nobody will ever miss it.