MKFirmware/MarlinKimbra #192: Extra end stops pins

coolerooney, 2019-05-20:
Hi

The MK4Duo fork of`Marlin supports extra endstops for the extra motors

Where do I connect the wires to?

It in is Feature and in Configuration Cartesian tabs

Please be very clear in your reply, `i am a real Noob. winking smiley)

Thx

The Noob

MagoKimbra, 2019-05-20:
Extra endstop for double engines in X or Y or Z?

coolerooney, 2019-05-21:
For all

Thx

MagoKimbra, 2019-05-21:
Ok in configuration_feature.h you can found:

```
//#define X_TWO_STEPPER_DRIVERS
//#define X_TWO_ENDSTOPS
//#define Y_TWO_STEPPER_DRIVERS
//#define Y_TWO_ENDSTOPS
//#define Z_TWO_STEPPER_DRIVERS
//#define Z_TWO_ENDSTOPS

```
And in configuration_pin you can set pin for extra endstop

```
#define X2_MIN_PIN          NoPin
#define Y2_MIN_PIN          NoPin
#define Z2_MIN_PIN          NoPin
#define Z3_MIN_PIN          NoPin
#define X2_MAX_PIN          NoPin
#define Y2_MAX_PIN          NoPin
#define Z2_MAX_PIN          NoPin
#define Z3_MAX_PIN          NoPin
```

coolerooney, 2019-05-21:
Hi

Yes i have found that
But what is the nr of the pin and where is this on the ramps / megatronics board?
Where do i connect the wires?

Thx

MagoKimbra, 2019-05-21:
where you want. Or use endstop pins that you don't use, such as Xmin and not Xmax, you can use the X_max pin for X2_min.

coolerooney, 2019-05-21:
Hi I am not an expert or anything close

please show me how to do
which pin nrs, where entered in de program, and how do the wires connect

apologies for being a Noob

best

carel

MagoKimbra, 2019-05-21:
Ok, but you have to explain to me what you need ..
What endstop do you use for the home?
Which engines do you have double? X or Y or Z?
On the ramps you have only one driver available for a double engine, for those who use it?

coolerooney, 2019-05-22:
Hi
Iam using the toolchain in a non 3d printer setup

But to keep it simple- a double x motor+driver, and want end stops on both, min and max
The min endstops is home

How do I achieve this- how do i assign a pin, which pin is available, and where do i find the connector on the board?

Thx again

Best

MagoKimbra, 2019-05-22:
Ok, so you have two X engines with two drivers, the home is on min. This is your configuration.
The firmware for 3D printers use endstops only during the homing phase, so the X MAX endstop is useless.
Anyway even if you want to use it if you want to put the double endstop X2_MIN you still have Y_MAX and Z_MAX free. So it's enough that you set in configuration_pins.h
#define X2_MIN_PIN ORIG_Y_MAX_PIN for example if you want to use the Y_MAX pin.

But you should use the online configurator to do this is much more practical and faster ...

coolerooney, 2019-05-22:
Agreed
How do I do this, see previous mail
The online configurator does not tell me which pins and where

Thx

MagoKimbra, 2019-05-22:
But the pins you have to choose them, there are no pins on the board for extra endstops ... So when you activate the double driver for X and activate the double endstop it asks you which pin to use and you put one of the endstops that you don't use ... Y_MAX or Z_MAX.

coolerooney, 2019-05-22:
Ok
But in de pins.h tab I cannot which pins are used......
And when it says NoPin, how do i give the right format No1 or 1 or?
Can i use the aux pins, for instance, but how do,i now what their pin nr is

MagoKimbra, 2019-05-22:
Ok, unfortunately we can't understand ...
If you configure everything online with the online configurator you don't have to touch anything in the files, you just have to choose the pin to use for X2_MIN_PIN, and I told you to use one of the endstop pins you don't use, in the configurator in the X2_MIN_PIN box put Y_MAX_PIN or Z_MAX_PIN that you don't use ... If you want to use another pin on the AUX connector and you need to know what pin is what number it is, I can't know it, you download the datasheet of your card with all the numbering and see which one is the pin ...

coolerooney, 2019-05-22:
Sorry for my Noob-ness

Found the configurator, looks good, by when 2 carriages are selected, it only gives you an additional min stop, no additional end stop- how do i.....

Thx

Best

MagoKimbra, 2019-05-22:
[image]

coolerooney, 2019-06-11:
Hi

Thank you for your patience with me ;)
Next question- how do I add the max endstop?

Thx again for your help- left a positive comment on your responses on reprap.org<http://reprap.org> forum;)

Carel

[endstop]<https://user-images.githubusercontent.com/7785361/58203511-89e96580-7cda-11e9-8ca0-77e0d6d892fb.jpg>