MKFirmware/MarlinKimbra #193: Nozzle Clean options

gearsawe, 2019-06-02:
Nozzle clean feature currently moves the Z position which is not needed in a system where the bed  moves up and down. Is it possible to assign in the Configurations_Fearture.h
#define NOZZLE_CLEAN_END_POINT   {100, 60, (Z_MIN_POS + 1)}
to replace the Z_MIN_POS with something like **current Z Poistion** with maybe an offset?

MagoKimbra, 2019-06-02:
But isn't the cleaning point on the plan?

gearsawe, 2019-06-02:
My printer works like that of a Fortus printing system. My bed only moves up and down. no need to move the bed to a certain z position. The default Z_MIN_POS would cause the head to crash into what ever was printing. I would like to move the  bed down 1mm from the current position. Like retraction. Then go to a cleaning position in the XY only.
I can probably do this all with a tools change script in Simplify3D. Just thought it might simplify things to have the printer handle the cleaning process rather than scripting all the movement in simplify.